SuperHouse/esp-open-rtos #125: Ending a FreeRTOS task crash

pellepl, 2016-04-09:
I might be out on deep waters here, but it seems that if I write a task that do not ```while forever``` but actually exits, I get a fatal exception 20, which might be stemming from ```esp-open-rtos/FreeRTOS/Source/tasks.c:2373```.

I made an one-liner modification of the [examples/simple/simple.c:14](https://github.com/SuperHouse/esp-open-rtos/blob/master/examples/simple/simple.c#L14), from

```
    uint32_t count = 0;
    while(1) {
        vTaskDelay(100);
```

to 

```
    uint32_t count = 0;
    while(count < 10) {
        vTaskDelay(100);
```

After 10 seconds, I get:

```
Fatal exception (20): 
epc1=0x00000000
epc2=0x00000000
epc3=0x40225b9d
excvaddr=0x00000000
depc=0x00000000
excsave1=0x00000000
```

addrsourcing epc3 which seems to be the only interesting address but might just be from a previous context switch, I get:

```
0x40225b9d is in prvIdleTask (/home/petera/proj/wisleep/esp-open-rtos/FreeRTOS/Source/tasks.c:2133).
2128	
2129				A critical region is not required here as we are just reading from
2130				the list, and an occasional incorrect value will not matter.  If
2131				the ready list at the idle priority contains more than one task
2132				then a task other than the idle task is ready to execute. */
2133				if( listCURRENT_LIST_LENGTH( &( pxReadyTasksLists[ tskIDLE_PRIORITY ] ) ) > ( unsigned portBASE_TYPE ) 1 )
2134				{
2135					taskYIELD();
2136				}
2137			}
Line 2133 of "/home/petera/proj/wisleep/esp-open-rtos/FreeRTOS/Source/tasks.c" starts at address 0x40225b9d <prvIdleTask+125> and ends at 0x40225ba5 <prvIdleTask+133>.
Dump of assembler code for function prvIdleTask:
   0x40225b20 <+0>:	addi	a1, a1, -32
   0x40225b23 <+3>:	s32i.n	a12, a1, 24
   0x40225b25 <+5>:	s32i.n	a13, a1, 20
   0x40225b27 <+7>:	s32i	a14, a1, 16
   0x40225b2a <+10>:	l32r	a13, 0x40225988
   0x40225b2d <+13>:	l32r	a12, 0x40225ac0
   0x40225b30 <+16>:	l32r	a14, 0x40225784
   0x40225b33 <+19>:	s32i.n	a0, a1, 28
   0x40225b35 <+21>:	s32i	a15, a1, 12
   0x40225b38 <+24>:	memw
   0x40225b3b <+27>:	l32i.n	a2, a13, 0
   0x40225b3d <+29>:	beqz	a2, 0x40225b9d <prvIdleTask+125>
   0x40225b40 <+32>:	memw
   0x40225b43 <+35>:	l32i.n	a2, a12, 0
   0x40225b45 <+37>:	l32i.n	a15, a14, 0
   0x40225b47 <+39>:	addi.n	a2, a2, 1
   0x40225b49 <+41>:	memw
   0x40225b4c <+44>:	s32i.n	a2, a12, 0
   0x40225b4e <+46>:	l32r	a0, 0x40225584
   0x40225b51 <+49>:	callx0	a0
   0x40225b54 <+52>:	beqz	a15, 0x40225b38 <prvIdleTask+24>
   0x40225b57 <+55>:	l32r	a0, 0x40225400
   0x40225b5a <+58>:	callx0	a0
   0x40225b5d <+61>:	l32i.n	a2, a14, 12
   0x40225b5f <+63>:	l32i.n	a15, a2, 12
   0x40225b61 <+65>:	addi.n	a2, a15, 4
   0x40225b63 <+67>:	l32r	a0, 0x4022598c
   0x40225b66 <+70>:	callx0	a0
   0x40225b69 <+73>:	l32r	a3, 0x4022576c
   0x40225b6c <+76>:	memw
   0x40225b6f <+79>:	l32i.n	a2, a3, 0
   0x40225b71 <+81>:	addi.n	a2, a2, -1
   0x40225b73 <+83>:	memw
   0x40225b76 <+86>:	s32i.n	a2, a3, 0
   0x40225b78 <+88>:	memw
   0x40225b7b <+91>:	l32i.n	a2, a13, 0
   0x40225b7d <+93>:	addi.n	a2, a2, -1
   0x40225b7f <+95>:	memw
   0x40225b82 <+98>:	s32i.n	a2, a13, 0
   0x40225b84 <+100>:	l32r	a0, 0x4022540c
   0x40225b87 <+103>:	callx0	a0
   0x40225b8a <+106>:	l32i.n	a2, a15, 48
   0x40225b8c <+108>:	l32r	a0, 0x40225494
   0x40225b8f <+111>:	callx0	a0
   0x40225b92 <+114>:	mov.n	a2, a15
   0x40225b94 <+116>:	l32r	a0, 0x40225494
   0x40225b97 <+119>:	callx0	a0
   0x40225b9a <+122>:	j	0x40225b38 <prvIdleTask+24>
   0x40225b9d <+125>:	l32r	a3, 0x40225774
   0x40225ba0 <+128>:	l32i.n	a2, a3, 0
   0x40225ba2 <+130>:	bltui	a2, 2, 0x40225b38 <prvIdleTask+24>
   0x40225ba5 <+133>:	movi.n	a2, 1
   0x40225ba7 <+135>:	l32r	a0, 0x40225408
   0x40225baa <+138>:	callx0	a0
   0x40225bad <+141>:	j	0x40225b38 <prvIdleTask+24>
End of assembler dump.
```

Now, I am a mere grasshopper when it comes to FreeRTOS, but shouldn't it suppose to handle exiting tasks? Or am I just exiting it the wrong way?

projectgus, 2016-04-10:
Hi pellepl,

It seems a bit counter-intuitive, but you're not really supposed to let the task function exit in FreeRTOS. There's a bit about in the FreeRTOS task docs here: http://www.freertos.org/implementing-a-FreeRTOS-task.html

If you want a task whose main function exists, the solution is to call `vTaskDelete(NULL)` at the bottom of the function. Keep in mind that having short-lived tasks will churn memory quite a bit, so you may want to consider finding ways to reuse them - or use the FreeRTOS timer features as another way to have asynchronous events (they all run in the context of a single 'timer' task).

I'm going to close this issue, but feel free to reopen if vTaskDelete(NULL) doesn't work as expected, or you think something else here is not quite right.

closed 2016-04-10 by projectgus

pellepl, 2016-04-10:
Thanks! I suspected so. Totally missed that in the freertos docs, sorry.