SuperHouse/esp-open-rtos #126: Automated build failing / toolchain options

projectgus, 2016-04-14:
As noted in #91, automated builds are currently failing as the esp-open-sdk toolchain revision we were using for [Travis - commit cd1d336](https://github.com/SuperHouse/esp-open-rtos/blob/master/.travis.yml#L4) - is no longer working. The esp-open-sdk commit is valid but the esptool fork had its history rewritten, so the submodule commit is no longer valid.

Also, recent revisions of esp-open-sdk don't finish building under `make STANDALONE=n`. Although the toolchain target is built, so for our purposes it is usable despite the error. However the maintainer suggests that STANDALONE=n is deprecated, so it may be going away entirely (see comments on [this PR](https://github.com/pfalcon/esp-open-sdk/pull/148)).

Currently contemplating best long term options for a toolchain (esp-alt-sdk, simple crosstool, something else). Suggestions welcome.

esp-alt-sdk is tempting because we can then support prebuilt toolchains, including for Windows, but they're on gcc 5.3.0 not 4.x which is a bit scary.

drasko, 2016-04-14:
@projectgus what do we need from this SDK? From what I understand only cross-toolchain is needed, alongside with `lx106-hal`. `lwIP` is replaced, so is axTLS.

I would say that it would be easier to try @jcmvbkbc's [crosstol-NG](https://github.com/jcmvbkbc/crosstool-NG/tree/feb1fb829d5cb4f8739e488cbd2f0b72304705bb), use only build tuned for `esp-open-rtos` project and build only for Linux.

If toolchain and SDK is needed on Windows maybe we can think how to Dockerize the whole shebang. I did one test here: https://github.com/projectiota/uiota-esp8266-docker. [Dockerfile](https://github.com/projectiota/uiota-esp8266-docker/blob/master/Dockerfile) downloads, compiles and install whole SDK, alongside with Micropython and Espruino. Then you can use a script like this one: https://github.com/projectiota/uiota-esp8266-docker/blob/master/compile.sh to compile your programs from your PC host.

There is one-click Docker tool installation on Win/MacOS today: https://kitematic.com/, and there can be a pre-build image on DockerHub, similar to this one: https://hub.docker.com/r/projectiota/uiota-esp8266-docker/, that you can simply pull automatically.

Sharing an SDK like this is really easy, we just have to see how to:
1. Tighten the Docker image - currently is huge due to the fact that it is based on [Debian](https://github.com/projectiota/uiota-esp8266-docker/blob/master/Dockerfile#L15) system. However, if some effort is put we can make a customized Buildroot image with just a few MB in size.
2. See how to mount user project dir from host PC into Docker container, and if this is not possible on Win - then copy the project dir prior to compilation, similar to [this](https://github.com/projectiota/uiota-esp8266-docker/blob/master/compile.sh#L11)

jcmvbkbc, 2016-04-14:
> esp-alt-sdk is tempting because we can then support prebuilt toolchains, including for Windows, but they're on gcc 5.3.0 not 4.x which is a bit scary

Current esp-open-sdk is at 4.8.2 and even proposed update to the crosstool-ng-1.22.x leaves it at 4.8.3 by default.

pfalcon, 2016-04-14:
> The esp-open-sdk commit is valid but the esptool fork had its history rewritten, so the submodule commit is no longer valid.

`git pull --rebase` . That should be default, but I don't recommend setting it in your git config, and instead keep typing by default - that's because it's frequent question, so the answer should be right at the fingertips.

Otherwise, for people who'are interested in playing with toolchains, I'd recommend joining @jcmvbkbc's effort of making sure that xtensa toolchain is supported well in upstream crostool-NG, rather than in gazillion of forks, see https://github.com/crosstool-ng/crosstool-ng/issues/243 (@jcmvbkbc knows better, but in my list, he pushed all needed for baremetal toolchain - it just needs wide testing).

As for esp-open-sdk, it's aim remains the same as at the time of the creation - to be known working, known stable toolchain built as much as possible from source. Until there's a good, wide testbase to validate a new toolchain against, upgrading toolchain version in esp-open-sdk is much lower priority than working on such testbase or on actual esp8266 issues. So, if you suddenly see an issue you didn't see during previous year, you can strike a toolchain upgrade of the list and go vary dozen other parameters you were changing during this time.

projectgus, 2016-04-14:
@drasko wrote:
> From what I understand only cross-toolchain is needed, alongside with lx106-hal. lwIP is replaced, so is axTLS.

Yes, I should have clarified this. You are totally correct, and we also provide our own libc (projectgus/newlib-xtensa) which is the upstream libc plus a single FreeRTOS locking patch.

libhal is really only used in a couple of places, and removing it was discussed in #66 and seems like a fine option. So with a small amount of work, all we'd need is a toolchain.

We also currently pull in esptool.py via esp-open-sdk, but I'm finally part-way through submitting the upstream esptool.py to pypi so it will be pip-installable soon.

@jcmvbkbc wrote:
> > esp-alt-sdk is tempting because we can then support prebuilt toolchains, including for Windows, but they're on gcc 5.3.0 not 4.x which is a bit scary
> Current esp-open-sdk is at 4.8.2 and even proposed update to the crosstool-ng-1.22.x leaves it at 4.8.5 by default. 

Thanks Max, noted. I think people may have misread me here - I meant that I was scared by the prospect of jumping to a gcc 5.x toolchain for this project. I don't want to do that at this stage.

The part that appealed about esp-alt-sdk was being able to recommend a prebuilt toolchain for getting up and running quickly on common platforms (including a Windows option).

esp-alt-sdk also has some gcc 4.9 builds but I'm guessing these are also relatively untested on lx106 compared to 4.8.x.

@pfalcon wrote:
> git pull --rebase . That should be default, but I don't recommend setting it in your git config, and instead keep typing by default - that's because it's frequent question, so the answer should be [...]

I don't think that's a fix for this. The automated build broke when Travis did a clean clone trying to get a preset esp-open-sdk revision. The steps it was running were:

```
git clone --recursive https://github.com/pfalcon/esp-open-sdk.git
cd esp-open-sdk
git reset --hard cd1d336
```

pfalcon/esp-open-sdk@cd1d336 checks out, but that commit has a submodule pointing to pfalcon/esptool-py@aafef9ceb which no longer exists since history was rewritten there (github seems to have cached it but I can't resolve it via the git command line.)

(To give a full explanation of what the Travis script does, it doesn't build its own toolchain every time. It caches the toolchain unless you change the preset esp-open-sdk revision. However the Travis server sometimes flushes its cache.)

That specific problem is worked around for now with a97fb75 - the esp-open-sdk commit is updated, but it only builds a subset of the targets in order to work around pfalcon/esp-open-sdk#138 & pfalcon/esp-open-sdk#148.

> I'd recommend joining @jcmvbkbc's effort of making sure that xtensa toolchain is supported well in upstream crostool-NG

This might be the best choice if `STANDALONE=n` is going away. The requirement for this project is a stable toolchain that isn't too complicated for people to build on most platforms. With prebuilt options being a nice-to-have.

...

I also just now remembered that back in December I fixed the esp-open-rtos makefile to work with STANDALONE=y as well (e264d63). I had clean forgotten about that! So we have another option - continue to recommend esp-open-sdk, and just ignore the other libraries it installs into its sysroot.

I don't really want to recommend that option though. On a technical level, we can override the search paths but we can't prevent the linker from resolving the Espressif SDK libraries (or @pfalcon's lwip) if nothing else matches. On a philosophical level, it seems at cross-purposes for this project to recommend people install the Espressif SDK into their toolchain sysroot.

pfalcon, 2016-04-15:
> The automated build broke when Travis did a clean clone trying to get a preset esp-open-sdk revision.

Ah, I see. That indeed might not work, as a revision may go away. But I do try to keep master clonable/buildable all the time and ready to process any bugreports on that (for default "make") with high priority.

> but that commit has a submodule pointing to pfalcon/esptool-py@aafef9c which no longer exists since history was rewritten there

Ok, I switched to esp-open-sdk to esptool upstream, based on all the feedback recently received.

> To give a full explanation of what the Travis script does, it doesn't build its own toolchain every time. It caches the toolchain

That's neat, I need to look how it's done, we need to have similar setup for micropython (or maybe it should be done at esp-open-sdk, and let anyone download the latest build).

> This might be the best choice if STANDALONE=n is going away

It's not going away, it just requires some ifdef on Makefile level which everyone is lazy to do, including me, as I thought nobody really uses STANDALONE=n, completely forgetting about esp-open-rtos. I'm going to fix it, but working together on upstream crosstool-NG is still better option than munging forks around (and I'm sad I didn't have much progress with that, and that's why I'm drawing attention to it hoping that other folks can join @jcmvbkbc , there, in upstream).

pfalcon, 2016-04-15:
>  I'm going to fix it

Done.

> (or @pfalcon's lwip)

lwip built from source is in -llwip_open. It doesn't include proprietary espconn crap. Yes, eventually I'd like to drop proprietary -llwip and use built from source instead, but apparently that doesn't happen before there will be enough open-source projects which use open-source lwIP API and not proprietary espconn crap.

projectgus, 2016-04-15:
Thanks @pfalcon, much appreciated. I saw that commit so I've swapped Travis back in de4465b.

I'm going to close this issue for now - if STANDALONE=n isn't deprecated then we can continue as we were.

I like the idea of directly supporting crosstool-ng as well, though. Once #66 is in I might look into that, and also whether we can host prebuilt crosstool toolchains somewhere.

closed 2016-04-15 by projectgus