SuperHouse/esp-open-rtos #127: Cannot set  gpio 16 in esp03 as output.

hetii, 2016-04-16:
Hello ;)

In esp8266-03 is not possible by `gpio_enable(16, GPIO_OUTPUT);` set gpio 16 as output.

According to this [link](https://github.com/willemwouters/ESP8266/blob/master/sdk/esp_iot_sdk_v0.9.3/IoT_Demo/driver/gpio16.c) I was able to set it by adding:

`WRITE_PERI_REG(PAD_XPD_DCDC_CONF, (READ_PERI_REG(PAD_XPD_DCDC_CONF) & 0xffffffbc) | (uint32)0x1);   // mux configuration for XPD_DCDC to output rtc_gpio0
    WRITE_PERI_REG(RTC_GPIO_CONF, (READ_PERI_REG(RTC_GPIO_CONF) & (uint32)0xfffffffe) | (uint32)0x0);   //mux configuration for out enable
    WRITE_PERI_REG(RTC_GPIO_ENABLE, (READ_PERI_REG(RTC_GPIO_ENABLE) & (uint32)0xfffffffe) | (uint32)0x1);   //out enable
`

projectgus, 2016-04-17:
Hi @hetii,

Thanks for sending the bug report. You're correct that GPIO 16 is "special" and support is still pending in the esp-open-rtos core, there's an issue for that already - #60 .

I can't explain the crash you're seeing, if you use the [steps shown here on the wiki](https://github.com/SuperHouse/esp-open-rtos/wiki/Crash-Dumps) for decoding the value of epc1 in the crash dump, what do you see?

Angus

pellepl, 2016-04-18:
@hetii,

the crash is due to the watchdog - see the log, ends with ```wdt reset```. Try putting a ```WDT.FEED = WDT_FEED_MAGIC;``` inside you while forever loop (which I guess is experimental anyway).

Cheers

hetii, 2016-04-18:
@projectgus 
Its a bit odd, cause when I just set gpio16 as output esp03 have trouble to even boot,
after few restarts here is the full log:

```
a lot junks...
���3fff5988, prio:12, stack:200

ESP-Open-SDK ver: 0.0.1 compiled @ Apr 12 2016 23:47:34
phy ver: 273, pp ver: 8.3

SDK version:0.9.9
mode : sta(18:fe:34:9e:dc:cb)
add if0

Starting main loop!
Fatal exception (9): 
epc1=0x40104f7d
epc2=0x00000000
epc3=0x40225be0
excvaddr=0x01040001
depc=0x00000000
excsave1=0x3feffe00

 ets Jan  8 2013,rst cause:1, boot mode:(3,6)

load 0x40100000, len 31096, room 16 
tail 8
chksum 0x2e
load 0x3ffe8000, len 2168, room 0 
tail 8
chksum 0xa0
load 0x3ffe8880, len 5860, room 0 
tail 4
chksum 0x63
csum 0x63

a lot junks...
����3fff5988, prio:12, stack:200

ESP-Open-SDK ver: 0.0.1@������ed @ Apr 12 2016 23:47:34
phy ver: 273, pp ver: 8.3

SDK version:0.9.9
mode : sta(18:fe:34:9e:dc:cb)
add if0

Starting main loop!
Fatal exception (0): 
epc1=0x40225bdd
epc2=0x40100070
epc3=0x40244c97
excvaddr=0x00000220
depc=0x40225b30
excsave1=0x401013d2

 ets Jan  8 2013,rst cause:1, boot mode:(3,6)

load 0x40100000, len 31096, room 16 
tail 8
chksum 0x2e
load 0x3ffe8000, len 2168, room 0 
tail 8
chksum 0xa0
load 0x3ffe8880, len 5860, room 0 
tail 4
chksum 0x63
csum 0x63

a lot junks...
}����3fff5988, prio:12, stack:200

ESP-Open-SDK ver: 0.0.1@������ed @ Apr 12 2016 23:47:34
phy ver: 273, pp ver: 8.3

SDK version:0.9.9
mode : sta(18:fe:34:9e:dc:cb)
add if0

Startin
 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x40100000, len 31096, room 16 
tail 8
chksum 0x2e
load 0x3ffe8000, len 2168, room 0 
tail 8
chksum 0xa0
load 0x3ffe8880, len 5860, room 0 
tail 4
chksum 0x63
csum 0x63

a lot junks...
���3fff5988, prio:12, stack:200

ESP-Open-SDK ver: 0.0.1 compiled @ Apr 12 2016 23:47:34
phy ver: 273, pp ver: 8.3

SDK version:0.9.9
mode : sta(18:fe:34:9e:dc:cb)
add if0

StartiFFa
```

In most case I see  `rst cause:4` but sometimes also 1 or 20.

Here is the log from using xtensa-lx106-elf-addr2line to find a line that cause the exception:

```
 xtensa-lx106-elf-addr2line -e /home/hetii/workspace/elektronics/esp-open-rtos/examples/espstk/build/espstk.out 0x40104f7d
??:?
xtensa-lx106-elf-addr2line -e /home/hetii/workspace/elektronics/esp-open-rtos/examples/espstk/build/espstk.out 0x40104f7d
??:?
xtensa-lx106-elf-addr2line -e /home/hetii/workspace/elektronics/esp-open-rtos/examples/espstk/build/espstk.out 0x40225be0
/home/hetii/workspace/elektronics/esp-open-rtos/FreeRTOS/Source/tasks.c:1083
xtensa-lx106-elf-addr2line -e /home/hetii/workspace/elektronics/esp-open-rtos/examples/espstk/build/espstk.out 0x40225bdd
/home/hetii/workspace/elektronics/esp-open-rtos/FreeRTOS/Source/tasks.c:1083
xtensa-lx106-elf-addr2line -e /home/hetii/workspace/elektronics/esp-open-rtos/examples/espstk/build/espstk.out 0x40244c97
??:?
xtensa-lx106-elf-addr2line -e /home/hetii/workspace/elektronics/esp-open-rtos/examples/espstk/build/espstk.out 0x40100070
exception_vectors.o:?
```
Line 1083 of tasks.c is end of `INCLUDE_vTaskSuspend ` macro.

@pellepl I try your way with WDT.FEED but without any positive results.

Cheers

projectgus, 2016-04-19:
That's weird, @hetii . Are you able to post a full program (as minimal as possible, but compileable and runnable) that crashes, please?

@pellepl, thanks for taking a look at this. On the non-RTOS SDK, running in a while(1) loop is a recipe for watchdog problems. In an RTOS environment multitasking takes care of the problem, so you shouldn't hit the watchdog normally unless you disable interrupts or run a very high priority task that never calls vTaskDelay() (thereby starving out all lower priority taks). I don't think that's happening here.

hetii, 2016-04-21:
Welcome back after few days ;)

My issue is probably related to hardware. I change LDO and it seams that my board is much more stable now, Moreover I have probably some shortcut between esp module and my pcb, cause my ceramic antenna sometime is quite hot. 

I also find my other board with esp03 and there my software works fine.

@projectgus I think we can close this issue and regarding gpio 16 and bug #60 I think that end user should use the same api, even if other registers are used by it. If not then at least some warnings should be raised if someone use `gpio_enable(16, GPIO_OUTPUT);`

projectgus, 2016-05-16:
Thanks @hetii , closing in favour of #60.

closed 2016-05-16 by projectgus