SuperHouse/esp-open-rtos #128: Simple_cplusplus with rboot crashes on boot

mikuslaw, 2016-04-18:
Hi,

first I would like to thank you for your work. It looks great and it brings joy of writing a code on ESP :).

But back to the point. I tried to run simple_cplusplus example with OTA enabled. After building rboot with big flash support and flashing, I flash the firmware (with OTA enabled). 
On boot I get messages that tasks are created, I get information about static object being created and then I get crash dump with exception 2 from non-code address 0x3FFF1F60 or something similar. Of course after getting that address to address to line tool it gives me error.

I will add the console output when I'm back at home. 

Two things get the example to work: 
- building without OTA - works as expected with static object created 
- removing static object

I will try to debug it further, but maybe you have some idea what's different with OTA that would break the static object? The fact that I see the printf from constructor would suggest that static constructors are called on start. Maybe something related to stack when constructor is called?

Thanks for your work and any help,
Jerzy

mikuslaw, 2016-04-18:
I now see that example/cpp_01_tasks has OTA disabled in local makefile. I understand that it's some known limitation?

mikuslaw, 2016-04-18:
As promissed, this is the output I get when running OTA enabled simple_cplusplus:

pp_task_hdl : 3fff21b8, prio:14, stack:512
pm_task_hdl : 3fff2b88, prio:1, stack:176
Counter initialised with count 99
Fatal exception (2): 
epc1=0x3fff1f80
epc2=0x00000000
epc3=0x00000000
excvaddr=0x3fff1f80
depc=0x00000000
excsave1=0x3fff1f80

[mikuslaw@t420s simple_cplusplus]$ ./../../utils/addrsource 0x3fff1f80
No source file for address 0x3fff1f80.
No line number information available for address 0x3fff1f80
No function contains specified address

As far as I gathered the 0x3fff1f80 is in RAM? Also what is interesting, I can get the example to work by removing the line with printf from the constructor.

This really starts to look like we don't have enough stack to run printf in static object constructor. In what context the constructor is called?

projectgus, 2016-04-19:
Thanks for the comprehensive bug report. You're right that 0x3fff1f80 is in data RAM, that's very odd that the PC is jumping there.

I don't think the OTA=0 is relevant here, I think the person who submitted those examples just copied one which had OTA=1 in it originally. I'll remove that from the Makefile to avoid confusion.

I think you're correct that it's something todo with static constructors. These use some clues embedded by the linker script. Maybe either the clues or the code which calls them is getting broken.

The constructors should be embedded between the "magic symbols" __init_array_start and __init_array_end, here:
https://github.com/SuperHouse/esp-open-rtos/blob/master/ld/common.ld#L182

... and then the code which calls them is here:
https://github.com/SuperHouse/esp-open-rtos/blob/master/core/sdk_compat.c#L26

I'm guessing somehow a symbol with an address in RAM (0x3fff1f80) is getting embedded into the region that should just be constructor function addresses, and then getting called.

It's weird that it only happens in OTA mode, because this part of the linker script is the same in both cases. The only bits that are different between OTA and non-OTA is the first part of the linker script with starting offsets.

You can use objdump (and possibly the .map file in the build directory) to see which symbols are being linked between __init_array_start & __init_array_end, this might give you some clues.

mikuslaw, 2016-04-19:
I will provide the map file when I'm back home today. 

One more thing: When I remove the printf form the constructor, the static object is constructed correctly and the example works. I was thinking about the situation when the static constructor is called in context without proper amount of stack. The simple constructor (without printf) can fit in a very limited space on stack ( lr/pc/some regs). When we add the printf to the mix, the requirement increases drastically. In that case, it's possible that we overwrite the LR with some data pointer used in printf. Then we just jump back to that address. 

I will try to get some idea about where the stack is located before the FreeRtos starts in these two cases (ota, nonota).

mikuslaw, 2016-04-19:
I attach the map file. On the line 5460 you will find that at 0x3fff1f50 starts the heap. Could be that printf overwrites a stack (located on the heap of course) of another task.
[simple.map.txt](https://github.com/SuperHouse/esp-open-rtos/files/226896/simple.map.txt)

projectgus, 2016-04-19:
Can you try checking out the "fix/static_ctor" branch with that fix please? It fixes the problem for me in non-OTA configuration (which also crashed for me) but I don't have time to test OTA.

(As it turns ou that fix was already committed on the `feature/tests` branch, I just cherry-picked it into master.)

mikuslaw, 2016-04-20:
Sure, I will test it today after getting from work. I was testing stability of a project of mine overnight and I didn't want to stop the test. 

I wonder who sets the stack pointer on boot. I would imagine that it would be a first stage boot. Or maybe like cortex m it just loads the pointer to stack on reset. 

I will try the fix and let you know. I guess that answering my question would require jtag. Maybe a project for some long night session.

projectgus, 2016-04-20:
Great, please let me know how you get on.

Regarding the stack pointer, as it happens I did some fishing around about it the other day, I left some notes in the feature/test branch as well:
https://github.com/SuperHouse/esp-open-rtos/blob/feature/tests/ld/common.ld#L235

(I really need to finish that branch so I can merge it!)

The stack pointer comes to us in sdk_user_start already initialised by the ROM intiialisation routines, which use the stack as well. However ideally there shouldn't be anything left on the stack at that point, because obviously we're not going to return to the ROM code. However 0x3ffffce9 is 791 bytes below the end of RAM - I don't know if that memory is reserved for some system purpose, or if we can use it ourselves if we just push the stack pointer a bit higher.

(At this stage there are no tasks, the task stacks come from the heap but this "supervisor" stack just comes from the top of RAM.)

mikuslaw, 2016-04-20:
Ok, I tested the fix you sent. 

The good thing is that it doesn't crash. I still have a bit funny behavior. The output of printf in static constructor is garbage:

ets Jan  8 2013,rst cause:2, boot mode:(3,4)

load 0x40100000, len 1344, room 16 
tail 0
chksum 0x9c
load 0x3ffe8000, len 660, room 8 
tail 12
chksum 0xb8
csum 0xb8

rBoot v1.3.0 - richardaburton@gmail.com
Flash Size:   32 Mbit
Flash Mode:   QIO
Flash Speed:  40 MHz
rBoot Option: Big flash

Booting rom 0.
Count��Poz���m��z�h��Qh���pp_task_hdl : 3fff2190, prio:14, stack:512
pm_task_hdl : 3fff2b60, prio:1, stack:176
frc2_timer_task_hdl:3fff5618, prio:12, stack:200

ESP-Open-SDK ver: 0.0.1 compiled @ Apr 20 2016 23:20:07
phy ver: 273, pp ver: 8.3

SDK version:0.9.9
mode : sta(5c:cf:7f:02:f4:27)
add if0
Counter initialised with count 12
Counter initialised with count 24
local counter 13 static counter 99 newly allocated counter 24
local counter 13 static counter 99 newly allocated counter 25
scandone
local counter 13 static counter 99 newly allocated counter 26

projectgus, 2016-04-20:
Ah, I think I know why. The clock is still running at 52MHz not 80MHz when the constructor execute, so the baud rate is briefly wrong.

Please try with the commit I just pushed.

mikuslaw, 2016-04-21:
Thank you very much for all the effort that you put into that. 

I disabled in the user code, the function to change the baudrate to 115200, so I'm running it at 74880 all the time (at least from the user perspective). That would mean that after putting the printf data into the send buffer, the baudrate changed briefly, as you mentioned, to a different value, then back to 74880 - beginning of the message is ok, middle is garbled, the rest of log is ok). 

I will confirm that with logic analyzer, and will test the fix. Thanks.

projectgus, 2016-04-21:
No problems. Keen to hear how the fix goes, I think the situation was exactly like you describe.

It might be a good idea for us to add a call to uart_tx_flush() inside uart_set_baud(), I can't think of any circumstances where you'd want to change baud rate midway through transmitting characters - and much more likely it'll catch people out like this.

(Although that's actually only part of the story, I've seen elsewhere that flushing the FIFO only flushes until the last byte has left the FIFO buffer, not until the last byte has been sent out on the serial port. So calling uart_tx_flush() / uart_set_baud() immediately can still corrupt the last byte, unless you also delay one byte period in between them to let the final byte leave the port. It's tricky!)

foogod, 2016-04-21:
Yeah, that's a good point.. perhaps we should change uart_tx_flush() to
make sure it waits long enough for the last byte to leave over the wire
before returning, too..

mikuslaw, 2016-04-22:
I tested first the output without the fix. As you suggested the serial switches for some short moment to 115200 and then back to 74880. The data that was sent both on 74880 and 115200 is correct.

After pulling the fix, everything is as expected. 

```
ets Jan  8 2013,rst cause:1, boot mode:(3,0)

load 0x40100000, len 1344, room 16 
tail 0
chksum 0x9c
load 0x3ffe8000, len 660, room 8 
tail 12
chksum 0xb8
csum 0xb8

rBoot v1.3.0 - richardaburton@gmail.com
Flash Size:   32 Mbit
Flash Mode:   QIO
Flash Speed:  40 MHz
rBoot Option: Big flash

Booting rom 0.
pp_task_hdl : 3fff2190, prio:14, stack:512
pm_task_hdl : 3fff2b60, prio:1, stack:176
Counter initialised with count 99
frc2_timer_task_hdl:3fff5618, prio:12, stack:200

ESP-Open-SDK ver: 0.0.1 compiled @ Apr 22 2016 18:29:56
phy ver: 273, pp ver: 8.3

SDK version:0.9.9
mode : sta(5c:cf:7f:02:f4:27)
add if0
Counter initialised with count 12
Counter initialised with count 24
local counter 13 static counter 99 newly allocated counter 24
local counter 13 static counter 99 newly allocated counter 25
scandone
local counter 13 static counter 99 newly allocated counter 26
local counter 14 static counter 99 newly allocated counter 26
add 0
aid 4
cnt 

```

Thank you very much for your help. I would say that this issue was resolved.

One more question, I noticed that you, projectgus, were the person that did the changes to openocd to support esp. Can you just tell me if using that version of openocd and the gdb from esp-open-sdk is the way to get it running?

closed 2016-04-22 by mikuslaw

projectgus, 2016-04-24:
Thanks for confirming. Fix pushed!

> One more question, I noticed that you, projectgus, were the person that did the changes to openocd to support esp. Can you just tell me if using that version of openocd and the gdb from esp-open-sdk is the way to get it running?

Yes, I wrote the original support then someone else cleaning it up a bit, then a third party posted it to the openocd gerrit tracker here: http://openocd.zylin.com/#/c/3348/

I haven't tested this newer version, but I'd recommend using that version of openocd not any other. The gdb from esp-open-sdk should work fine. Some earlier versions the stack traces were corrupt (otherwise worked fine), but I think esp-open-sdk has an updated version now.

projectgus, 2016-04-24:
(If you do use that version of openocd and it works for you, please comment on the Gerrit change! It'd be good to get some confirmations on there so it can potentially be merged.)