SuperHouse/esp-open-rtos #132: lwip_read() returns response headers... sometimes

jsuiker, 2016-04-25:
I have a feeling this might not be a bug, but I'd appreciate some help on it anyway.  I started playing around with the http_get example and tried hooking it up to a nodejs server I threw together.  I can't figure out why the socket is returning response headers from my nodejs server and not when I serve the same page with Apache.
[apache_response.txt](https://github.com/SuperHouse/esp-open-rtos/files/233983/apache_response.txt)
[node_response.txt](https://github.com/SuperHouse/esp-open-rtos/files/233984/node_response.txt)
[request-response_headers.txt](https://github.com/SuperHouse/esp-open-rtos/files/233985/request-response_headers.txt)

foogod, 2016-04-25:
I believe the reason for this is twofold:

1. The http_get example actually uses the (very old) HTTP/0.9 protocol to talk to the remote server
2. Your nodejs server is not technically fully HTTP compliant, because it does not properly support HTTP/0.9 requests.

The original HTTP protocol (which later became "version 0.9") actually had no concept of headers.  The client simply sent a "GET _path_" command, and the server returned the content, with no metadata or headers of any kind.  Later, folks realized this was really not flexible enough for a lot of things, they added the concept of client and server headers as part of HTTP/1.0.  To make a HTTP/1.0 request, the client would change the GET (or POST, etc) request to have the form "GET _path_ HTTP/1.0" instead.  In order to support backwards compatibility, however, if a server received a request without the "HTTP/1.0" on the end, it would still return the response (with no headers) the old HTTP/0.9 way.

I just looked at the `http_get` example code, and it's just sending "GET _path_" to the server, which means it's sending an old-style HTTP/0.9 request.  Apache is fully backwards-compatible, so it actually still supports this, and returns a response with no headers as it should.  It appears your node.js server, on the other hand, does not actually know how to send back a HTTP/0.9 response, so it sends back a HTTP/1.x response, even though the client asked for a HTTP/0.9 one (this is arguably wrong, but nowadays HTTP/0.9 is so old it probably doesn't matter much to most people).

Arguably, I think we really should change the `http_get` example to use a HTTP version from this century (which would also fix your issue), so I think it's valid to have this as an open bug against esp-open-rtos..  If you'd be interested in fixing it up we'd welcome a submission, otherwise I'll see if I can look into it sometime soon..

foogod, 2016-04-25:
I've updated the title of this issue to better reflect the actual problem..  I also just realized when looking at the code again that the GET request that `http_get` sends is not only using HTTP/0.9, but is actually technically an invalid HTTP/0.9 request (it sends a "User-Agent" header, but headers aren't allowed in HTTP/0.9 requests), so this definitely needs a little work..

jsuiker, 2016-04-25:
Yea I've already done some reworking on the http_get example so I'll get it up to date and submit a PR for it.  Thanks for the insight, I was really banging my head against the wall on this one :)

RLuckom, 2016-04-26:
@jsuiker I was running into the same problem--couldn't get an expressjs server to process a POST request. I eventually came up with https://github.com/RLuckom/esp-open-rtos/blob/dht11/examples/dht11/dht11.c#L88 . It's very ugly (I started out just trying to read a DHT11 sensor and then just kept globbing more stuff into that file), but it works.

jsuiker, 2016-04-28:
I think we'll just continue conversation over on issue #133

closed 2016-04-28 by jsuiker