SuperHouse/esp-open-rtos #133: HTTP client library

projectgus, 2016-04-26:
This is related to #132, but I thought it'd warrant its own issue to discuss/track.

For a long time I've wanted to include an HTTP client library in esp-open-rtos. Making raw socket requests works for simple actions (apart from obvious problems like #132), but it'd be good to have some wrappers for common HTTP operations.

* I tried porting a stripped down libcurl, but even stripped down it's not suitable for this size of processor
* mbed has its own C++ HTTPClient library (although I'm not sure where the canonical source for it is), but it's in C++ and fairly mbed-centric so it may not be suitable.

Most people seem to just DIY on raw sockets for the subset of HTTP they need. We could write our own I guess, it might be of wider interest if it only depends on LWIP.

foogod, 2016-04-26:
Heh, that's funny.. I was just thinking of creating a ticket for this as well.  I agree that this would be a very useful general-purpose module to have in esp-open-rtos (and I also agree that an embedded-friendly C-only HTTP client library could be useful in general)..

This might not be a bad project for somebody new to esp-open-rtos, as it shouldn't require much beyond learning the ins and outs of the LWIP stuff (and familiarizing onesself with the HTTP protocol, of course).  I'm sorta envisioning this as something that could potentially be built in several stages:

1) Really minimal, but still useful:
* Supports GET requests
* Supports basic HTTP/1.0 protocol
* Just call the API routine and get back the payload (HTML page or whatever).

2) Basic two-way capabilities:
* Supports POST requests
* Support for sending/receiving custom HTTP headers
* Possibly more of a callback-based interface which could then be wrapped by simpler APIs

3) Reasonably full-featured basic HTTP client
* Reporting/custom handling of HTTP status codes
* Auto-handling HTTP redirects, etc
* HTTP basic auth support
* Other less common request types (HEAD, PUT, ...), get-if-modified, etc.
* Maybe extra routines to help with URL encoding, query strings, etc.
* Basic HTTPS support

4) All kinds of more advanced features (most probably all optional add-ons, if they ever are implemented):
* Cookie handling
* Support HTTP/1.1 and chunked encoding
* Streaming data to/from server
* Compressed payloads
* Establishing websockets
* etc

jsuiker, 2016-04-28:
So I've started working in this direction and want to see what you guys think of what I've come up with.  As a rough start, I moved a lot of the http request-building and -executing to an 'extras' library which interacts with the lwip API in much the same way as the http_get example did.  From what I can tell, building requests is the easy part, but efficiently parsing responses in a resource-constrained environment will be the difficult hurdle to overcome.  I came across a project called [http_parser](https://github.com/nodejs/http-parser) that is apparently very light-weight and doesn't require any external libraries.

I'm just poking around with it at the moment, but have had a bit of [trouble](https://github.com/jsuiker/esp-open-rtos/commit/bf49a10b07a8af95a8d1a55674093c04d91bbebc)(see below) compiling it, which I think has to do with the esp-open-rtos version of libc.  Eventually, I would love to figure out a way to get http_parser talking with the lwip 'socket' since this seems to be the least taxing on memory resources.  I realize that the traditional socket_t is NOT the same as the lwip_sock object so it may require significant customization to the http_parser code to do so.

In any case, I'd appreciate some input on the issue I'm struggling with now, which is the compilation errors I'm receiving while making the reworked http_get example.
```
make: Entering directory `/opt/SuperHouse/esp-open-rtos/examples/http_get'
CC /opt/SuperHouse/esp-open-rtos/extras/http/http.c
CC /opt/SuperHouse/esp-open-rtos/extras/http/http_parser/http_parser.c
AR build/http.a
LD build/http_get.out
../../libc/xtensa-lx106-elf/lib/libc.a(lib_a-abort.o):(.literal+0x0): undefined reference to `_exit'
../../libc/xtensa-lx106-elf/lib/libc.a(lib_a-abort.o): In function `abort':
/home/gus/dev/esp/rtos/newlib/build/xtensa-lx106-elf/newlib/libc/stdlib/../../../../../newlib/libc/stdlib/abort.c:63: undefined reference to `_exit'
../../libc/xtensa-lx106-elf/lib/libc.a(lib_a-signal.o):(.literal+0x0): undefined reference to `_getpid_r'
../../libc/xtensa-lx106-elf/lib/libc.a(lib_a-signal.o):(.literal+0x4): undefined reference to `_kill_r'
../../libc/xtensa-lx106-elf/lib/libc.a(lib_a-signal.o): In function `_raise_r':
/home/gus/dev/esp/rtos/newlib/build/xtensa-lx106-elf/newlib/libc/signal/../../../../../newlib/libc/signal/signal.c:163: undefined reference to `_getpid_r'
/home/gus/dev/esp/rtos/newlib/build/xtensa-lx106-elf/newlib/libc/signal/../../../../../newlib/libc/signal/signal.c:163: undefined reference to `_kill_r'
collect2: error: ld returned 1 exit status
make: *** [build/http_get.out] Error 1
```

projectgus, 2016-05-03:
Hi @jsuiker,

Sorry for the slow reply. This sounds like awesome news!

node-js's http-parser looks like it could be perfect, nice find.

The errors are because we don't have an abort() syscall implementation in esp-open-rtos at the moment. Issues #54 tracks it. I'm actually working on something else right now that I can dovetail with adding abort, give me a day or two and I'll send a PR that adds it in.

However, looking through the http_parser source it looks like it's use of assert.h that triggers abort (if assertions fail). If you add -DNDEBUG to your CFLAGS then this will disable assertions and remove the reference to `abort()`, so it should compile OK as-is. (see https://github.com/SuperHouse/esp-open-rtos/blob/master/libc/xtensa-lx106-elf/include/assert.h)

jsuiker, 2016-05-19:
I had some time to work on this today, wondering if you two could provide a bit of feedback.  I'm somewhere between Stage 2 and Stage 4 as outlined by @foogod... but as I said before, my C skills are pretty elementary.  By the way, should I specifically mention you guys in these notes or do you get notified of their presence anyway?  Seems my last one took a while to get picked up - no worries though.

Thanks for the commits @projectgus, it is nice to have everything working as it should.  It seems you added some crash dump features as well?  I'm curious if there is any documentation to help a youngster like me interpret the hieroglyphs.  I'm running into the following crash after several minutes of 'GET'ing an html page:

```
Fatal exception (3): 
epc1=0x4010635d
epc2=0x00000000
epc3=0x40229708
excvaddr=0x400000b4
depc=0x00000000
excsave1=0x3fff4e40

Stack: SP=0x3fff4e40
0x3fff4e40: 00000077 000005ac 00000077 00000000
0x3fff4e50: 00000000 3fff5a70 3fffef6e 000005ac
0x3fff4e60: 00000000 00000000 00000000 00000001
0x3fff4e70: 3fff3120 00000077 00000078 4023061d
0x3fff4e80: 3ffeec1c 00000008 0000003a 00000000
0x3fff4e90: 00000000 00000077 3fffdf30 40231057
0x3fff4ea0: 00000060 3ffeb06c 3fff33e4 4022e888
0x3fff4eb0: 3fff3120 3fff4ee0 3fffdf30 00000000

Free Heap: -1124
_heap_start 0x3fff3698 brk 0x400005ec supervisor sp 0x3ffffae0 sp-brk -2828 bytes
arena (total_size) 53076 fordblks (free_size) 1704 uordblocks (used_size) 51372
 ets Jan  8 2013,rst cause:4, boot mode:(1,6)
wdt reset
 ets Jan  8 2013,rst cause:4, boot mode:(1,6)
wdt reset
```

rodrigopex, 2016-07-11:
Just trying to help, I don't know if there is still time to that. This link would be helpful http://www.demailly.com/~dl/wwwtools.html#http-tiny.

kanflo, 2016-08-04:
Hi @jsuiker, did you make any progress on debugging the crash?

kanflo, 2016-08-04:
@rodrigopex, thanks for the links. Mongoose seems very interesting but cannot be used because of licensing.

jsuiker, 2016-08-04:
I haven't looked at it since... I got a 'real job' in the mean time where I have to code all day, so coming home to work on this is a bit of a drag for me now...

furkanmustafa, 2016-09-28:
I'm currently working on one (http client library). But I had too many crashes with lwip so far.

First thing helped with crashes was increasing `TCPIP_THREAD_STACKSIZE` in lwipopts.h to 1024.

I was trying so hard to find mistakes in my code until I figured this out.

malachib, 2017-06-06:
I'm curious has anyone gotten mongoose to work in this environment?  I already understand licensing for it is restrictive.