SuperHouse/esp-open-rtos #140: Clients disassociated every 10 seconds in SoftAP mode

projectgus, 2016-05-09:
(Placeholder as I'm about to send an email to the mailing list with details)

ourairquality, 2016-09-15:
Took a look at this, and thanks for all the analysis above. Found that `hostap_input` does update slot 0x108 but the context passed to `hostap_handle_timer` does not match the context it is writing into.

Quick look suggests that the problem is that the callback create with xTimerCreate receives a handle, and not the expected argument. Is it practical to implement every callback in C to change their handlers, or do we need a trampoline callback for handle this compatibility issue for now?

This might also explain some of the crashes in the timer paths.

chennakeshavaSoC, 2016-09-21:
Is there any update on this 10 seconds timeout and max connection!

ourairquality, 2016-09-21:
@chennakeshavaSoC It is fixed by https://github.com/SuperHouse/esp-open-rtos/issues/22 It has not been merged yet, but close when it has been.

chennakeshavaSoC, 2016-09-21:
Thanks for quick response, I mean 
Clients disassociated every 10 seconds in SoftAP mode #140

Is there any updates on #140

chennakeshavaSoC, 2016-09-21:
Can you please share us the fix code, I would like to use this functionality its a critical update for us.

ourairquality, 2016-09-21:
@chennakeshavaSoC Sorry, copy typo, see https://github.com/SuperHouse/esp-open-rtos/issues/221 for the patch.

closed 2016-09-22 by kanflo

rooi-oog, 2016-10-18:
I can't realize was this fixed or not? Because this issue still applying to me. Clients still disassociating. But not prints "max connection". And not after 10 seconds but 20.

phlemoine, 2016-10-18:
It's working for me with the recent fix that got merged in master branch.
See #221

rooi-oog, 2016-10-18:
I cloned repo 2 days ago. I just watched - yes that patch is present. But not working for me.

ourairquality, 2016-10-19:
@rooi-oog If no one can reproduce this then it might be up to you to try to narrow it down. In case it helps please consider https://github.com/SuperHouse/esp-open-rtos/pull/224 which added open implementations of some function in these code paths. For example this allows you to instrument `sdk_hostap_handle_timer` to see if it is disassociating due to a timeout, and `sdk_cnx_node_search` was a good spot to dump the state.

rooi-oog, 2016-10-19:
I just tried the workaround (sdk_hostap_handle_timer) and it works for me.