SuperHouse/esp-open-rtos #143: Example http_get_mbedtls crash

Terstegge, 2016-05-27:
Hello,

I was playing around with some examples and noticed that http_get_mbedtls crashes in my environment:

```
0350: 03 00 00 00 00 00 00 00 29 c0 00 00 3c ce 27 40
0360: 05 00 00 00 06 00 00 00 09 00 00 0 03 00 00 00

boot info:
0000: 7f 00 01 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e
0010: 0f 10 11 12 13 14 15 16 17 18 19 7f
�s{�c�������pocl���g{s�c�Ǝp����gs쎟��p��c�oхͭ}�������ɥ��х��MA=���M-ٕɍ�������5�����ٕ���ٕ�M-ٕ�ͥ���mode : sta(18:fe:34:f1:2e:65)
add if0
HTTP get task starting...

. Seeding the random number generator... ok
. Loading the CA root certificate ... failed
!  mbedtls_x509_crt_parse returned -0x2180

abort() invoked at 0x402261a9.

0x402261a9: http_get_task at /usr/local/git/esp-open-rtos/examples/http_get_mbedtls/http_get_mbedtls.c:132

Stack: SP=0x3fff79d0
0x3fff79d0: 00000000 00000000 402805a0 402261ac

0x402805a0: sdk_rc4_skip at ??:?

0x402261ac: http_get_task at /usr/local/git/esp-open-rtos/examples/http_get_mbedtls/http_get_mbedtls.c:135

0x3fff79e0: a5a5a5a5 a5a5a5a5 a5a5a5a5 a5a5a5a5

Free Heap: 29384
_heap_start 0x3fff2918 brk 0x3fff8844 supervisor sp 0x3ffffae0 sp-brk 29340 bytes
arena (total_size) 24364 fordblks (free_size) 44 uordblocks (used_size) 24320

ets Jan  8 2013,rst cause:2, boot mode:(1,6)

ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset
```

Any idea what went wrong?
Thanks for the nice work so far - Andreas

projectgus, 2016-05-27:
Hi Andreas, 

Thanks for raising this issue. Are there any modifications to the example source apart from changing ssid_config.h? What hardware (in particular flash size) are you running on?

Angus

Terstegge, 2016-05-27:
Hi Angus,
I have a 512kb flash on a handy 2,54mm spacing board:
[http://shop.in-circuit.de/product_info.php?cPath=21_51&products_id=67](http://shop.in-circuit.de/product_info.php?cPath=21_51&products_id=67)

I made a fresh installation of esp_free_sdk and esp_free_rtos this morning, and only have modified the ssid_config.h file.
In other examples the connection to my DSL router was working (http_get without SSL was working without problems).

projectgus, 2016-05-28:
Strange. The http_get_mbedtls binary image does come close to 512kB, but it should still have some headroom.

Are you passing FLASH_SIZE=4 (for 4mbit) to make somehow? You can pass it on the command line, in a local.mk file, or you can edit the default.) The default size is 16mbit (2MB flash), and some data is stored by the SDK at the end of flash, so its possible that going past the end makes something go wrong.

I can't really explain it otherwise - the error result suggests that mbedtls is failing to parse the compiled-in certificate data, but the same certificate data loads fine for me. :/

The breakout board you have looks a little light on powerrail capacitance, maybe it's browning out and the flash is mis-reading? But again that seems unlikely to me.

Good luck puzzling it through!

Terstegge, 2016-05-29:
Passing the FLASH_SIZE is a good point! But rebuilding and running with FLASH_SIZE=4 (I also noticed the -fs 4m option in esptool) did not help :(
I am pretty sure it is a flash size problem, tough. Unfortunately a have no 16mbit flash lying around to solder it on my board. So I will probably order/wait for another board with bigger flash or really get a bigger flash chip and replace the current one.
Power supply should be okay. I have a large capacity on the power rail to reduce the spikes on the 3.3V line while WLAN sending, and the LDO I am connected to should deliver 500mA (TPS73533 on a MSP432 Launchpad), which should be sufficient (I hope...).

Terstegge, 2016-06-10:
A got a ESP12e with 4 MiB Flash, and now the example works without problems -> issue can be closed, its only a flash size issue (maybe worth a comment in the example code)...

closed 2016-06-10 by Terstegge

MightyPork, 2016-06-18:
@Terstegge may I ask where you got the ESP from, and does it really have 4 megs?
I've always lived under the belief that ESP12E's have 1 MiB

Terstegge, 2016-06-19:
Sure you may :) I ordered a standard ESP-12e module from ama**ne (I guess if you search the term 'ESP-12e' you will surely find one).
Looking at the schematics (e.g. [http://www.esp8266.com/wiki/doku.php?id=esp8266-module-family#esp-12-e_q](http://www.esp8266.com/wiki/doku.php?id=esp8266-module-family#esp-12-e_q)), you will see that the 12e module has a 32Mibit serial flash, which in fact is 4MiB.