SuperHouse/esp-open-rtos #147: No effect when changing configCPU_CLOCK_HZ or configTICK_RATE_HZ

Zaltora, 2016-06-06:
Hi, I tried to change some settings of FreeRTOSConfig.h for time control and see what happen.
This is my code "tick.cpp" : 

```
#include "FreeRTOSConfig.h"
#include "espressif/esp_common.h"
#include "esp/uart.h"
#include "FreeRTOS.h"
#include "task.h"
#include "esp8266.h"

void handler_task_loop (void *pvParameters)
{

    uint32_t time = sdk_system_get_time() ;
    uint32_t time_rtos = xTaskGetTickCount()*portTICK_RATE_MS ;

	while(1) {

	printf("Uptime_rtos:%u ms\t\t",xTaskGetTickCount()*portTICK_RATE_MS);
	printf("Uptime_sdk:%u ms\n",sdk_system_get_time()/1000);
	printf("loop_rtos:%u ms\t\t",(xTaskGetTickCount()*portTICK_RATE_MS)-time_rtos);
	printf("loop_sdk:%u ms\n",(sdk_system_get_time()-time)/1000);
	printf("------------------------------\n");
	time_rtos = xTaskGetTickCount()*portTICK_RATE_MS ;
	time = sdk_system_get_time() ;
	vTaskDelay(100/portTICK_RATE_MS);
	}
}

extern "C" void  user_init(void)
{
	uart_set_baud(0, 115200);
	printf("SDK version:%s\n", sdk_system_get_sdk_version());

	sdk_system_update_cpu_freq(SYS_CPU_160MHZ) ;

	printf("CPU freq:%u\n", sdk_system_get_cpu_freq());
	printf("CPU clock:%u\n", configCPU_CLOCK_HZ);
	printf("TICK RATE:%u\n", configTICK_RATE_HZ);
	printf("------------------------------\n");

    xTaskCreate(handler_task_loop, (signed char *)"loop", 256, NULL, 3, NULL);
}
```

when i set these settings  in FreeRTOSConfig.h file : 

```
#define configTICK_RATE_HZ			( ( portTickType ) 1000 )
#define configCPU_CLOCK_HZ			( ( unsigned long ) 160000000 )
```
this is my output:

```
SDK version:0.9.9
CPU freq:160
CPU clock:160000000
TICK RATE:1000
------------------------------
mode : softAP(1a:fe:34:de:a0:cf)
add if1
bcn 100
Uptime_rtos:3 ms		Uptime_sdk:253 ms
loop_rtos:1 ms		loop_sdk:2 ms
------------------------------
Uptime_rtos:105 ms		Uptime_sdk:758 ms
loop_rtos:100 ms		loop_sdk:499 ms
------------------------------
Uptime_rtos:205 ms		Uptime_sdk:1258 ms
loop_rtos:100 ms		loop_sdk:499 ms
------------------------------
Uptime_rtos:305 ms		Uptime_sdk:1758 ms
loop_rtos:100 ms		loop_sdk:499 ms
------------------------------
Uptime_rtos:405 ms		Uptime_sdk:2258 ms
loop_rtos:100 ms		loop_sdk:499 ms
------------------------------
Uptime_rtos:505 ms		Uptime_sdk:2758 ms
loop_rtos:100 ms		loop_sdk:499 ms
------------------------------
```
Real loop is 500 ms  when tick calculations said 100 ms. Only default setting do exact time calculation.
I forget to change some setting ,Error in my code    ?
ty for help

projectgus, 2016-06-29:
Hi @Zaltora ,

Thanks for opening this issue. The "tick timer" (CCOUNT) is currently initialised in the binary SDK's libmain, so the timing is hardcoded for the default settings.

Once the "open-libmain" branch is merged we can fix this properly:
https://github.com/SuperHouse/esp-open-rtos/blob/open-libmain/open_esplibs/libmain/os_cpu_a.c#L66

Otherwise, if you want run at 160MHz with predictable task timing then you can use
```
#define configTICK_RATE_HZ ( ( portTickType ) 200 )
```
... which just happens to work because you're doubling the tick rate. This is of course not the long term solution, the long-term solution is to merge open-libmain so we can configure this properly.

Angus

projectgus, 2016-07-02:
Hi @Zaltora, if you have a chance then you can you please try the open-libmain branch in #154 and let me know if you get the expected timing behaviour.

I left some notes in commit 5ad2a3579 that some behaviour from inside the binary SDK may change if you change the tick period. Once this change is merged I'll copy those notes it to the wiki somewhere. However it seems like most functionality still works as expected with 1000Hz ticks and this change.

Zaltora, 2016-07-04:
Hi @projectgus  , ty for your reply  and your help . I was expected this part was not open source since i asked to official forum :(
But i didn't expect reverse engineering, awesome.
I will test if it is work.

commit 9511d0c3 (2016-06-29): Honour values of configCPU_CLOCK_HZ & configTICK_RATE_HZ for tick rate

Fixes #147

* Can vary tick rate from 100Hz via configTICK_RATE_HZ. Note that the
  SDK binary libraries are hard-coded to assume the tick rate is 100Hz,
  so changing the tick rate may have unexpected consequences for lower
  layer WiFi behaviour (such as certain kinds of timeouts happening
  faster/slower.)

* Setting configCPU_CLOCK_HZ to 160MHz means ESP will set 160MHz during
  initialisation. Only 80MHz and 160MHz are supported.

* Timing of tasks is no longer affected by current CPU freq (whether set
  via configCPU [... cut]
commit 5ad2a357 (2016-06-29): Honour values of configCPU_CLOCK_HZ & configTICK_RATE_HZ for tick rate

Fixes #147

* Can vary tick rate from 100Hz via configTICK_RATE_HZ. Note that the
  SDK binary libraries are hard-coded to assume the tick rate is 100Hz,
  so changing the tick rate may have unexpected consequences for lower
  layer WiFi behaviour (such as certain kinds of timeouts happening
  faster/slower.)

* Setting configCPU_CLOCK_HZ to 160MHz means ESP will set 160MHz during
  initialisation. Only 80MHz and 160MHz are supported.

* Timing of tasks is no longer affected by current CPU freq (whether set
  via configCPU [... cut]
commit 2345dc51 (2016-07-06): Honour values of configCPU_CLOCK_HZ & configTICK_RATE_HZ for tick rate

Fixes #147

* Can vary tick rate from 100Hz via configTICK_RATE_HZ. Note that the
  SDK binary libraries are hard-coded to assume the tick rate is 100Hz,
  so changing the tick rate may have unexpected consequences for lower
  layer WiFi behaviour (such as certain kinds of timeouts happening
  faster/slower.)

* Setting configCPU_CLOCK_HZ to 160MHz means ESP will set 160MHz during
  initialisation. Only 80MHz and 160MHz are supported.

* Timing of tasks is no longer affected by current CPU freq (whether set
  via configCPU [... cut]
commit 678b59ba (2016-07-06): Honour values of configCPU_CLOCK_HZ & configTICK_RATE_HZ for tick rate

Fixes #147

* Can vary tick rate from 100Hz via configTICK_RATE_HZ. Note that the
  SDK binary libraries are hard-coded to assume the tick rate is 100Hz,
  so changing the tick rate may have unexpected consequences for lower
  layer WiFi behaviour (such as certain kinds of timeouts happening
  faster/slower.)

* Setting configCPU_CLOCK_HZ to 160MHz means ESP will set 160MHz during
  initialisation. Only 80MHz and 160MHz are supported.

* Timing of tasks is no longer affected by current CPU freq (whether set
  via configCPU [... cut]
closed 2016-07-13 by projectgus