SuperHouse/esp-open-rtos #151: implementation issue with LWIP and FreeRTOS

DutchFox2, 2016-06-27:
I'm encountering problems with LWIP in esp-open-rtos.
Easiest steps to reproduce:
- start with the http_get example from the esp-open-rtos examples directory,
- modify the URL, so it will download a somewhat larger file (100 kB should do),
- in the receive loop, do not print the full received data, but just a '.' or something,
- put a vTaskDelay within the receive loop (100-1000 msec)

This last step is crucial. It mimics an application where the received data can only be processed slowly (in my case this will be a slow serial link to an even slower secondary microcontroller). The TCP protocol should handle this gracefully, because the TCP Rx Window will/should only be updated when there is enough free memory available to allow for further reception.

Observed results: the download starts normally, but will hang after a while (ca 10 sec). The 'hang' will be short in the beginning, but subsequent hangs will take longer and longer, until the delays are getting so long that the webserver finally gives up and stops serving the rest of the download (we are talking _minutes_ already now).
In a Wireshark log, lots of requests for retransmits can be seen, which is an indication that a serious number of packets is getting lost in the ESP.

According to the [LWIP-user mailing list](https://lists.gnu.org/archive/html/lwip-users/2016-04/msg00019.html), the most likely cause is a faulty vendor implementation: threading issues, failing to fetch _all_ received packets after a NIC interrupt, etc.
I have tried to pinpoint this problem myself, but without success. It might even be possible that things are already going wrong in the closed-source libraries from Espressif.
Compared to the code I mentioned in the LWIP-user mailing list, I have already stripped down the situation to the almost unmodified http_get example (see above), which still displays the faulty behavior.

If I try an equally simple example from esp-open-**sdk** (also modified to include the delay in the receive loop), with exactly the same download URL, the download succeeds without problems! Same compiler toolchain, same webserver, same hardware. This leads me to the conclusion that the problem must be somewhere within esp-open-rtos.

It would be great if somebody could try to reproduce this issue.

projectgus, 2016-06-28:
Hi,

Thanks for reporting this. I can reproduce it, and I think I might have seen it ages ago when investigating #20. Back then I put it down to having to disable interrupts (the "delay" was caused by writing to flash), but it seems like that's not really a requirement to trigger this.

Playing around, a workaround seems to be shrinking the TCP receive window size. So if you add to your lwip/lwipopts.h:

```
#define TCP_WND (TCP_MSS * 2)
```

(the default value is TCP_MSS * 4)

... then this seems to mask the problem. I think retransmits still occur, but they're fast retransmits not slow retransmits so the long pauses don't happen and things generally appear smooth again.

I'm going to keep looking into this, but I thought I'd mention this workaround as it's probably enough to get you up and running again.

Angus

DutchFox2, 2016-07-19:
Hi, I'm glad that you can reproduce it. Thanks for stepping in and for mentioning a workaround. Lowering the TCP window indeed helps, I now have a fair chance that the ESP can actually download the file (was: 0%). This is (for now) enough to keep me going in the development of the software.
Hopefully you (or anybody else) will be able to find the real cause.

recursify, 2016-08-13:
I can confirm the issue with a file size of 400K... Changing TCP_WND does seem to help the overall throughput, but the stalls on the last chunk of data.

kanflo, 2016-08-15:
Changing TCP_WND makes miracles for me too when netcating an image to my [wifi display app](https://github.com/kanflo/eor-adafruit-ili9341/tree/master/example). Upload time drops from over 18 seconds worst case to 700ms for a 225kByte BMP.

Must try this on OTA mode that sometimes take forever...

phlemoine, 2016-09-01:
I can reproduce the issue even with smaller files than 100k ( around 35k I think, I'll update the thread when I narrow it down) . I have looked briefly at enabling lwip debug, but without success so far. I didn't find the uart redirect that the lwip forums are mentioning. Has anyone managed to enable these debug options?

recursify, 2016-09-01:
@phlemoine it's been a while since I enabled it, but was likely here https://github.com/SuperHouse/esp-open-rtos/blob/02c35d8a7153a9f8be895f6a170bccdc1307f0a0/lwip/include/lwipopts.h#L392

phlemoine, 2016-09-01:
yeah i got that, i also enabled SOCKETS_DEBUG and TCPIP_DEBUG to LWIP_DBG_ON
but i am not getting traces on the UART. i am not sure where the UART is selected.

phlemoine, 2016-09-02:
Nvm, my lwipopts include was messed up. I am getting traces now

phlemoine, 2016-09-20:
i abandonned debugging this after i realized i wasnt building with the right lwipopts.h. with the suggested workaround to define TCP_WND to TC_MSS*2, i no longer have any issue in downloading large files . multiple 200kB over night are no longer an issue.

furkanmustafa, 2016-09-28:
I have also experienced all these problems. 

@phlemoine 
> i also enabled SOCKETS_DEBUG and TCPIP_DEBUG to LWIP_DBG_ON
but i am not getting traces on the UART. i am not sure where the UART is selected.

along with those flags, you also need to define `LWIP_DEBUG`. it writes to uart0 by default.

@recursify 
> but the stalls on the last chunk of data.

You might have solved this already as it's been a month already, but just in case this is still same, or just for future reference; it's very high likely to be caused by http keep-alive. lwip has no way to understand if there is no more data coming or not when the connection is still on. So if you include `Connection: close` in your *request* headers, the server side will probably behave and close the connection after streaming the file content.

phlemoine, 2016-10-01:
The lwipopts.h file must be in your current directory or at the location
it's coming from esp-open-rtos. If you're unsure of which lwipopts.h gets
built , add a #error pragma and make sure the build fails ... You'll know
which is used.
Once I figured this out, I got the traces on uart0 without any additional
changes

recursify, 2016-11-15:
Finally got around to firing up Wireshark to see what was going on.  I have a scenario where there are two simultaneous TCP streams: one websocket TCP stream with a small amount of back and forth data, and another one that is downloading a 400K file over HTTP.

Before I start the HTTP download, the websocket stream seems healthy in wireshark (no retransmits, no duplicate ACKs).  After the download begins, the websocket stream starts showing numerous retransmits and duplicate ACKs.  Not sure if that helps narrow down where the issue might be, but thought it was interesting that one stream was negatively affecting another.

@DutchFox2 - when you said that you got a minimal version to work in `esp-open-sdk` , what version were you using?  And was it using the same version of the Espressif provided binaries, or (more likely) a more recent one?

@projectgus (or others) - if you were to take an educated guess, where would think the issue is, the esp-open-rtos LWIP port or the Espressif provided binaries?  `sys_arch.c` and `esp_interface.c` look pretty straightforward so my hunch is the latter... but that's just a hunch :)