SuperHouse/esp-open-rtos #152: crash of the MQTT client when receiving "large" messages

Munrek, 2016-06-28:
MQTT client is disconnecting or crashing when receiving large messages, ( ~ more than 200, 250 bytes )

Details :

Fatal exception (9): 
epc1=0x40208957
epc2=0x00000000
epc3=0x40217c98
excvaddr=0x23232327
depc=0x00000000
excsave1=0x40208ace
Registers:
a0 40208ace a1 3fff9620 a2  3fff96a0 a3  23232323
a4  00000000 a5  00000000 a6  3fff2368 a7  00000000
a8  00000000 a9  3ffeaa68 a10 3ffea610 a11 3ffeaa44
a12 3fff97a8 a13 3fff96a0 SAR 00000020

Stack: SP=0x3fff9620
0x3fff9620: 00000000 3fff97a8 3fff9800 4020872c
0x3fff9630: 40208610 3fff96e3 3fff97a8 40208a68
0x3fff9640: 3fff9680 00000020 3fff12e0 00000000
0x3fff9650: 3fff96a0 3fff97a8 fffffffd 40208ace
0x3fff9660: 00000090 3ffea610 3ffeaa44 00000790
0x3fff9670: 3fff97a8 00000010 00000000 00000020
0x3fff9680: 4021588c 00000000 3fff96a0 4020872c
0x3fff9690: 00000010 3fff97a8 fffffffd 40208c27

Free Heap: 27012
_heap_start 0x3fff1418 brk 0x3fffa4d0 supervisor sp 0x3ffffae0 sp-brk 22032 bytes
arena (total_size) 37048 fordblks (free_size) 4980 uordblocks (used_size) 32068

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 2292, room 16 
tail 4

(gdb) list *0x40208957
0x40208957 is in readPacket (/home/theo/Dev/Embedded/esp8266/esp-open-rtos/extras/paho_mqtt_c/MQTTClient.c:92).
87	    int rem_len = 0;
88	
89	    /* 1. read the header byte.  This has the packet type in it */
90	    Network* ip = c->ipstack;
91	
92	    int (*mqttread) (Network*, unsigned char*, int, int) = ip->mqttread;
93	
94	    if ( mqttread(
95	        c->ipstack,
96	        c->readbuf,

baoshi, 2016-06-29:
Hello, I'm the one who wrote the Paho port.To be frank I did not test the large message case. Thanks for the catching. I'll look into this. Did you just publish the message through a mosquito server? I'm building a testing case here. 
Btw how you get gdb working?

Munrek, 2016-06-29:
Hi, yep the message came through a mosquitto server.

This is no live debugging, I loaded my binary with xtensa-thingy-gdb, and looked for the line of code that raised the exception.

I tried to make the same thing by disassembling with objdump  ( and mixing with the source code, with the option -S ), but quite an interesting bug happened : all the functions of MQTTClient.c where disassembled correctly but with no C code, whereas it worked fine for other files. Maybe this behavior in consistent with the possibly alignment problem ? I could not get objdump to get more verbose to know what was happening.

funnydog, 2016-07-04:
It's unrelated to this bugreport, but since the author of the paho_mqtt port is here I'd like to point out a possible bug in MQTTClient.c:

```
int  sendPacket(MQTTClient* c, int length, Timer* timer)
{
    int rc = FAILURE,
        sent = 0;

    while (sent < length && !expired(timer))
    {
        rc = c->ipstack->mqttwrite(c->ipstack, &c->buf[sent], length, left_ms(timer));
        if (rc < 0)  // there was an error writing the data
            break;
        sent += rc;
    }
    if (sent == length)
    {
        countdown(&(c->ping_timer), c->keepAliveInterval); // record the fact that we have successfully sent the packet
        rc = SUCCESS;
    }
    else
        rc = FAILURE;
    return rc;
}
```

in the while loop the code keeps sending the full buffer length, for each iteration, not taking into account that it already sent some data.

sheinz, 2016-11-25:
This seems to be resolved with https://github.com/SuperHouse/esp-open-rtos/pull/158

closed 2016-11-25 by sheinz