SuperHouse/esp-open-rtos #156: Dash "-" in SSID for wifiSSID cant connect

marclave, 2016-07-01:
Seems like if there is a "-" in the SSID it wont find the network? I am using the MQTT example, can anyone else confirm this?

kanflo, 2016-08-04:
Hi @marclave. Sorry for the lack of attention, I will try to reproduce this.

marclave, 2016-08-04:
@kanflo no worries!

marclave, 2016-08-06:
@kanflo did you have any luck reproducing this?

sheinz, 2016-08-07:
I tried **http_get** example with SSID containing a dash "-". Everything works fine in my case.
@marclave can you try **http_get** example with your AP?

kanflo, 2016-08-08:
@marclave I have not had the possibility to try. Went vacation with a dud FTDI-board meaning I can only dry code ESP stuff this week :-/

projectgus, 2016-08-09:
Silly question, but is it possible the SSID you're having problems with doesn't contain the ASCII dash but instead contains one of the other non-ASCII dash characters?
https://www.cs.tut.fi/~jkorpela/dashes.html

AndreaGreco, 2016-11-16:
SSID to my house is called `Wi-Fi2` and working fine over ESP8266

sheinz, 2016-11-25:
I will close this for now, if there's more information regarding this issue, please reopen it.

closed 2016-11-25 by sheinz