SuperHouse/esp-open-rtos #161: const data in RAM

SaimenSays, 2016-07-10:
In my case I've discovered that all `const` data is located to RAM per default. Due to RAM limitations on ESP I think it is more usefull to leave them in FLASH. Also if you are using external external repos it is nearly impossible to add `ICACHE_FLASH_ATTR` to every struct or string.

Copied to and unnecessary eating up RAM:
`const char test[] = "---test---";`
`printf("---test---")`

Stayes in FLASH, not decreasing heap:
`const char test[] ICACHE_RODATA_ATTR= "---test---";`

Why does a `const` is not residing in FLASH by default?
Is there maybe anycompiler option to do so?

kanflo, 2016-07-10:
Not having looked at it I would guess the linker file is to "blame". Placing const data in RAM would be a good thing though, unless one wants it in RAM for performance reasons.

SaimenSays, 2016-07-10:
@kanflo: I don't understand your answer? What do you mean with "blame" . I've read thast ESP has a internal fast responding cache for readin slow external flash. So performance should not really be affected by reading directly from readonly section.

projectgus, 2016-07-10:
Hi @SaimenSays,

Thanks for your interest in our project. :)

I think there might be some confusion here. Unlike Espressif's official RTOS SDK, const data in esp-open-rtos is stored in flash by default (since PR #142 in May). There were some technical complexities to doing this, #11 is a long discussion but it goes over most of them if you're interested.

The `ICACHE_RODATA_ATTR` attribute is not used on our platform.

To demonstrate, if I run `make size` on the blink example I get:

```
./build/blink.out  :
section            size         addr
.text             30117   1074790400
.data              2182   1073643520
.rodata            1967   1073645704
.bss              33608   1073647672
.irom0.text      198696   1075847184
```

Of these sections, .text is in IRAM (instruction RAM). .bss, .data and .rodata are in data RAM (some read-only data needs to be available in RAM), and .irom0.text is stored in flash.

If I then paste some code like this into a task:

```diff

--- a/examples/blink/blink.c
+++ b/examples/blink/blink.c
@@ -17,6 +17,7 @@ const int gpio = 2;
 void blinkenTask(void *pvParameters)
 {
     gpio_enable(gpio, GPIO_OUTPUT);
+    printf("All of the strings go here! Absolutely all of them!\n");
```

And re-run `make flash`:

```
./build/blink.out  :
section            size         addr
.text             30117   1074790400
.data              2182   1073643520
.rodata            1967   1073645704
.bss              33608   1073647672
.irom0.text      198776   1075847184
```

... the only size that changes is irom0.text, because the string is stored in flash.

If you're seeing size changes in .data and/or .rodata, or reduced heap size at runtime, then it might be because the data is being copied to somewhere non-const? If you have an exact example that you can show then it might help us figure it out.

projectgus, 2016-07-10:
Something I've meant to do for a while is document the differences between esp-open-rtos and the official SDK: https://github.com/SuperHouse/esp-open-rtos/wiki/ESP-SDK-Differences

SaimenSays, 2016-07-12:
Hello again!
@projectgus You are completely right! You give me the hint. After checking the `common.ld`, I recognized that my external repo was not correctly updated long time. Now all const is stored in .irom0.text! And now I have 16k more RAM free.

**Thanks for all your work on this project!**

closed 2016-07-12 by SaimenSays

projectgus, 2016-07-12:
Good to hear :)