SuperHouse/esp-open-rtos #162: After flashing TLS example no more flashable

jackasses, 2016-07-13:
Hello,

after i flash the TLS-Server example on the ESP i can't flash the ESP.

The example boot and work. The listening via UART is possible. All prints from the example are arriving.
Sending information via UART isn't possible or isn't processed.
The connection with esptool etc. isn't possible.

Is there a way to bring it to a mode were i can flash it?

best regards

projectgus, 2016-07-13:
Hi @jackasses,

The serial bootloader on the ESP is in hardware (the bootloader runs from ROM.) Going into program vs bootloader mode is determined by the pin configuration at reset or poweron.

There are some details about the pins involved here:
https://github.com/themadinventor/esptool#entering-the-bootloader

Angus

jackasses, 2016-07-15:
Thanks for this Details.
Now i know more this Problem.

I done everything witch this modul to flash it.
Now it is in the bootloader mode and sends with 74,8kBaud the boot informations.
That is the only thing it do.

projectgus, 2016-07-17:
If you're seeing boot information then you're correctly getting to the bootloader mode, so esptool.py should be able to do flashing operations, etc. 

If you pull gpio1 high and then reset (or power off & on) then the normal program will run instead.

I'm going to close this issue because it seems like esp-open-rtos is working properly. If you think there's a bug in esptool.py then feel free to open an issue there, however it sounds to me like if you follow the hardware troubleshooting steps on the esptool.py page then you will get it going OK.

closed 2016-07-17 by projectgus