SuperHouse/esp-open-rtos #172: Porting "constants in ROM" to esp-open-sdk

slaff, 2016-07-26:
I saw that your SDK puts automagically constants into ROM, thus saving RAM which is scarce. 
I would like to port this feature to esp-open-sdk (open version of the Espressif NON OS SDK).  And I am trying to figure out the minimal set of files that are involved and code that needs to be migrated. 
And I would need your help to tell me if that will work or not.

As far as I could figure out two files were mainly involved:
```
core/include/common_macros.h
core/exception_vectors.S
```

From `core/include/common_macros.h` I need probably only these lines
```
#ifdef	__cplusplus
    #define IROM __attribute__((section(".irom0.literal")))
#else
    #define IROM __attribute__((section(".irom0.literal"))) const
#endif
```

Concerning `core/exception_vectors.S` - can I just copy it completely ( together with `led_debug.s` ) and hope that it will work?

And finally - it there a way to instruct the compiler to use the ROM attribute for all constants instead of me putting them manually?

projectgus, 2016-07-26:
HI @slaff,

Interesting project. Unfortunately it would be a lot more work than this! I don't know exactly how much, but more.

The exception_vectors.S code implements the entire set of Xtensa CPU exception vectors. We can do this because we compile those parts of the code from source, but as esp-open-sdk uses Espressif's binary libraries for these parts you would need to replace the relevant parts of the Espressif SDK binary libraries.

The other parts of exception_vectors.S are RTOS specific, in that they control things like switching between tasks. The same vectors exist for the non-OS SDK, but they'll have different code in them.

The default placement of symbols in esp-open-rtos is different and this is determined by the linker script (in the ld directory). This is how the constant symbols (compiled into .rodata) end up in the irom section at link time, So you'd need to modify the Espressif SDK linker script in order to have these wind up in irom as well.

> And finally - it there a way to instruct the compiler to use the ROM attribute for all constants in a file instead of me putting them manually?

I don't know of a way to instruct the compiler, but you can add entries in the linker script file that match certain object file names and use these to pick up specific sections. For example the "/* libc also in IRAM */" section takes some libc functions in certain source files (in irom by default) and puts them in iram for performance reasons. You can do something like this for per-file relocation of constants.

Given the amount of work and reverse engineering involved, can I suggest investigating esp-open-rtos and seeing if it will meet your needs?

kanflo, 2016-07-27:
Hi @slaff. I would answer "ready" to that. I have a couple of nodes running EOR and pushing/fetching data via MQTT and they are rock solid.

closed 2016-11-10 by slaff