SuperHouse/esp-open-rtos #174: Swap in a faster memcpy

projectgus, 2016-07-27:
This is a bit "full circle", but:

* When we swapped in our own newlib for a libc we got this memcpy: https://github.com/projectgus/newlib-xtensa/blob/xtensa-2_2_0-lock/newlib/libc/machine/xtensa/memcpy.S - it does byte reads until a 4 byte aligned address, then word reads, and I think it does byte reads for any trailing bytes as well.

* With #142 we enabled the unaligned load handler for default constant data which was now in flash. So this memcpy got slower in more cases.

* In #163 @sheinz noticed substantial performance drop using the newlib memcpy for spiffs, and he's written a new one. He also points out that the one in Espressif's ROM is pretty fast.

So maybe we should swap out memcpy (and maybe some of the other very simple low level libc functions) to ones which never perform byte reads?

Could do this in one of several ways:
* Use the Espressif ROM binary-only implementations, swap them in at link time.
* Write our own and put them in our codebase, like @sheinz has done with his memcpy implementation. Swap them in at link time.
* Patch newlib-xtensa, given we already have some e-o-r specific modifications in there.

... I don't really mind which one. Leaving this issue open to track/discuss the desire for one of them, though!

sheinz, 2016-08-01:
> With #142 we enabled the unaligned load handler for default constant data which was now in flash. So this memcpy got slower in more cases.

Are you sure? Have you done any benchmarking? My preliminary testing shows no substantial performance difference between those functions when operating with RAM data buffers:
0x4000df48	memcpy
0x400018b4	ets_memcpy
0x400006c4	xthal_memcpy
newlib's memcpy

Am I correct that unaligned exception handler works only for unaligned flash(ROM) access?

>In #163 @sheinz noticed substantial performance drop using the newlib memcpy for spiffs, and he's written a new one. He also points out that the one in Espressif's ROM is pretty fast.

I couldn't use standard **memcpy** because it discards **volatile** qualifier for registers.
You might misunderstood me. I noticed performance drop between manually copying  data from registers into unaligned buffer with pure C. So, I had to use assembler to do it efficiently.

Doesn't e-o-r currently use [**memcpy** from rom](https://github.com/SuperHouse/esp-open-rtos/blob/master/ld/rom.ld#L62)?

sheinz, 2016-08-01:
I've done some benchmarking.
The number in the following log is the time in microseconds required to copy 128 bytes 1000 times.

RAM to RAM copy:

    newlib memcpy:
    aligned src, dst and size: 1777
    unaligned src, dst and size: 1826
    unaligned both src and size: 2274
    unaligned only src: 2275
    unaligned only dst: 2500
    unaligned only size: 1832
    unaligned both src and dst: 1939
    unaligned both dst and size: 2480

    rom ets_memcpy:
    aligned src, dst and size: 1907
    unaligned src, dst and size: 1941
    unaligned both src and size: 2400
    unaligned only src: 2402
    unaligned only dst: 2630
    unaligned only size: 1977
    unaligned both src and dst: 2007
    unaligned both dst and size: 2572

    rom memcpy:
    aligned src, dst and size: 1860
    unaligned src, dst and size: 1827
    unaligned both src and size: 2280
    unaligned only src: 2275
    unaligned only dst: 2505
    unaligned only size: 1826
    unaligned both src and dst: 1984
    unaligned both dst and size: 2477

    rom xthal_memcpy:
    aligned src, dst and size: 1875
    unaligned src, dst and size: 2543
    unaligned both src and size: 2456
    unaligned only src: 2419
    unaligned only dst: 3063
    unaligned only size: 1925
    unaligned both src and dst: 2602
    unaligned both dst and size: 3064

ROM to RAM copy:

    newlib memcpy:
    aligned src, dst and size: 3746
    unaligned src, dst and size: 5095
    unaligned both src and size: 5690
    unaligned only src: 4378
    unaligned only dst: 10052
    unaligned only size: 6645
    unaligned both src and dst: 8137
    unaligned both dst and size: 8506

    rom ets_memcpy:
    aligned src, dst and size: 4042
    unaligned src, dst and size: 5323
    unaligned both src and size: 5823
    unaligned only src: 4522
    unaligned only dst: 10199
    unaligned only size: 6740
    unaligned both src and dst: 8204
    unaligned both dst and size: 8751

    rom memcpy:
    aligned src, dst and size: 3745
    unaligned src, dst and size: 5098
    unaligned both src and size: 5694
    unaligned only src: 4380
    unaligned only dst: 10050
    unaligned only size: 6646
    unaligned both src and dst: 8138
    unaligned both dst and size: 8627

    rom xthal_memcpy:
    aligned src, dst and size: 3817
    unaligned src, dst and size: 4651
    unaligned both src and size: 4520
    unaligned only src: 4606
    unaligned only dst: 5296
    unaligned only size: 3984
    unaligned both src and dst: 4600
    unaligned both dst and size: 5299

The only implementation which doesn't throw unaligned exception interrupt is **xthal_memcpy**. Which is not a surprise as it is [designed to use only 32 bit access](https://github.com/tommie/lx106-hal/blob/master/src/memcopy.S#L61)
But it is slower when copying from RAM to RAM. So, it is a trade off.

projectgus, 2016-08-02:
Hi @sheinz,

I think you're right that I misunderstood the previous discussion, sorry. If everything you were doing was DRAM to DRAM then the unaligned exception handler never gets invoked, as data RAM is designed for byte-level access. I should have looked closer at the code first.

Is newlib's memcpy significantly slower if you use it in the spiffs implementation, in place of the hand-rolled one?

sheinz, 2016-08-02:
Hi Angus,

> Is newlib's memcpy significantly slower if you use it in the spiffs implementation, in place of the hand-rolled one?

Not at all. But as I already said standard memcpy ignores **volatile** qualifier. 

The code in Espressif rom uses **memw** instructions for each  load/store operation of SPI data registers. It's **volatile** implementation in C. So, I decided to do it as well.
I'm not sure what implications might be if not to use **memw** instruction for SPI data registers access.

My experiments show that **memcpy** that ignores **volatile** qualifier works just as good and as fast as my own implementation.

Maybe @foogod or @raburton might shed some light whether **memw** is really needed for SPI data registers access.

projectgus, 2016-08-05:
I'm not sure if MEMW is strictly necessary on LX106, but if it is needed then it's to ensure that any load/store operation ordered before MEMW completes fully before any operations ordered after MEMW are attempted. As I understand it, for LX106 this would only (if ever) be a risk for race conditions between operations on one slower bus and another faster bus.

In the case of memcpy_unaligned_XXX in spiffs it's copying in/out of the SPI data registers, so that's safe in this case - the SPI peripheral is all in one bus, so it's unlikely anything will happen faster than anything else (and also it wouldn't matter if it does, as they're just data registers).

If you want guaranteed safety, I believe you could add a single `__asm__ volatile("memw");` after the memcpy() call that writes to SPI memory, and the same before the memcpy that reads from SPI memory. This should take care of any potential races. I don't think this will turn out to be strictly necessary, but it doesn't really hurt either.

(Sorry again that I didn't look at the code earlier, I totally misunderstood the problem to begin with!)