SuperHouse/esp-open-rtos #175: Implement a way to know when module is connected to an AP

doragasu, 2016-08-03:
Up to now, to guess when the ESP in station mode is connected to an AP, I was polling sdk_wifi_station_get_connect_status() until I got the status STATION_GOT_IP. That works perfect when using DHCP, but now I have to use static IP... STATION_GOT_IP status is never trigged when not using DHCP, and there is no other code indicating the module is connected to the AP.

When using Espressif SDK, you can set an event handler callback and wait for the event EVENT_STAMODE_CONNECTED. AFAIK this cannot be done on esp-open-rtos, and there is no other substitute for this function.

Could it be possible to add a function that returns the connection status? I think this is a big problem if you need to use static IP (at least it is for me).

doragasu, 2016-08-03:
I have been doing some more tests and it looks like it works (I get STATION_GOT_IP status) when using static IP and configuring it inside user_init(). If I do it outside user_init(), status is set to STATION_CONNECTING and never changes. So maybe what we need is not a way of knowing when the AP is connected, but a wait of been able to change IP configuration outside user_init().

kanflo, 2016-08-04:
An event callback handler for wifi status would be a good addition to EOR (and also a way of querying connection status). If you have any suggestions I'd be happy to review them. Also if you find a way of changing IP config outside of user_init, you are more than welcome with a PR.

doragasu, 2016-08-04:
About the event callback handler, wasn't it removed from EOR in the first place? It's in the original SDK, so I thought it was removed maybe because the developers thought it was not really needed. But there's maybe other reason (maybe it did not work or interfere with other functions). It would be interesting to hear why it was removed from the project developers.

I suppose changing the IP outside user_init() requires fiddling with SDK internals (reverse engineering the ip/dhcp configuration routines). Unfortunately my reverse-engineering abilities aren't good enough for getting this done in a reasonable amount of time :(.

kanflo, 2016-08-04:
I see. @projectgus, can you shed some light on this?

projectgus, 2016-08-05:
I don't believe the Espressif SDK version we're using ever had wifi status callbacks, unfortunately.

Changing the IP outside user_init is probably possible using pure LWIP, without going down to the SDK layers. Something like:

```
struct netif *if = netif_find("en"); /* Name of the STA default interface */
netif_set_ipaddr(if, &new_addr);
```

If the ESP disconnects & reconnects to the AP then it will probably get set back to whatever is configured in the SDK layer, though.

I don't know what the "best" solution for knowing what the AP status is, though. A hacky way would be to hook all printfs() coming out of the SDK and look for the known status updates. A better way maybe to interface with the net80211 higher MAC functions that are in the binary libraries.

Will comment again if I think of anything else.