SuperHouse/esp-open-rtos #176: Function sdk_system_deep_sleep ignores argument.

sheinz, 2016-08-03:
From the function signature it is clear that system should go to deep sleep for the specified amount of microseconds. But regardless of specified value the following output is shown:
```deep sleep 1073s```

I verified that in function **sdk_system_deep_sleep** the value is correct.
And correct value is passed into 
``` sdk_os_timer_setfn(&sdk_sta_con_timer, _deep_sleep_phase2, (void *)time_in_us);```
But the timer callback function is called with incorrect value.

I checked with disabled **open_esplibs** the result is the same.

sheinz, 2016-08-03:
I've located the problem:
```
static void _deep_sleep_phase2(void *timer_arg) {
    uint32_t time_in_us = (uint32_t)timer_arg;

    printf("deep sleep %ds\n\n", time_in_us / 1000000);
```
The argument is not timer_arg but timer_handler. The actual argument should be extracted with function **pvTimerGetTimerID**:
```
    uint32_t time_in_us = (uint32_t)pvTimerGetTimerID(timer_handle);

    printf("deep sleep %ds\n\n", time_in_us / 1000000);
```

kanflo, 2016-08-04:
Nice catch, thanks. Could you please provide a PR with the fix?

sheinz, 2016-08-04:
I'm not sure if my fix is the best solution.
I looked into the new release of ESP8266_RTOS_SDK and they removed timers.o and use timers from ets_timer.o which does not use FreeRTOS timers.
If the timer from timers.o is used only once in function **sdk_system_deep_sleep** we might remove it as well and switch to the timer from ets_timer.o.
Less code is better.

sheinz, 2016-08-17:
As it turned out timers *sdk_os_timer_* are used by binary libraries as well as by *lwip* code.

If not properly extract a timer argument from the timer handle in the timer callback function the argument is invalid. 
For example currently in [lwip](https://github.com/SuperHouse/esp-lwip/blob/esp-1.4.1-rtos/espressif/espconn_tcp.c#L227) :
```
static void ICACHE_FLASH_ATTR
espconn_cclose_cb(void *arg)
{
	espconn_msg *pclose_cb = arg;
```
I'm not sure if timer argument is properly extracted in binary libraries... If not, passing argument to a timer callback function doesn't work at all.

The solution to this issue might be to rework the timers so the parameter in the callback function is an actual argument. Check each callback function in the binary libraries and verify that it doesn't convert a parameter of a callback function to an argument using *pvTimerGetTimerID*. If it does... think how to handle it.

I start to think that this timer argument issue was the motivation for Espressif to abandon FreeRTOS timers and switch to their own timers. Or maybe it was just easier for them to grab a code from non-rtos-sdk and put it into rtos-sdk...

sheinz, 2016-08-17:
I missed **ets_timer.o** in *libmain.a*. It provides timers that don't use FreeRTOS timers and I believe return actual argument in a timer callback function. It hasn't been reverse engineered in _open_esplib_ yet. 

It still strange why the SDK libraries use *ets_timer_* in some places and *os_timer_* in another.

The simple fix for this issue would be just to use *ets_timer_* in *sdk_system_deep_sleep*. I will check if it works I will provide a PR.

sheinz, 2016-08-19:
Fixed with https://github.com/SuperHouse/esp-open-rtos/pull/190

closed 2016-08-19 by sheinz