SuperHouse/esp-open-rtos #179: Function iomux_set_function doesn't work as expected.

sheinz, 2016-08-09:
From the function declaration 
`inline static void iomux_set_function(uint8_t iomux_num, uint32_t func)`
It seems like the second parameter *func* should be something like `IOMUX_GPIOn_FUNC_*` from iomux_regs.h. But it doesn't work. Because inside `iomux_set_function` *func* is converted with `IOMUX_FUNC(func)`. The GPIO function constants in *iomux_regs.h* already defined converted with `IOMUX_FUNC(func)`:

    #define IOMUX_GPIO2_FUNC_I2SO_WS           IOMUX_FUNC(1)
    #define IOMUX_GPIO2_FUNC_UART1_TXD         IOMUX_FUNC(2)
    ...

So, there's a double convention with `IOMUX_FUNC()`

The simple fix is just to remove `IOMUX_FUNC` in `iomux_set_function` which does work for me. If it is how it intended to work, I will provide a PR with the fix.

projectgus, 2016-08-10:
I think that's the correct solution. I was talking to @ourairquality about this issue a couple of weeks ago - https://github.com/SuperHouse/esp-open-rtos/pull/171#issuecomment-235264621

I think this is the correct change, as it keeps the register-only interface working correctly and makes it compatible with the iomux__xxx wrapper interfaces. I seem to remember there was a snag, but I can't think of it right now. Maybe @ourairquality has some comments to add.

projectgus, 2016-08-10:
I think I like your proposed solution here better than the one in #171, if there's no other downside.

sheinz, 2016-08-17:
PR https://github.com/SuperHouse/esp-open-rtos/pull/187 fixes the issue.

closed 2016-08-17 by sheinz