SuperHouse/esp-open-rtos #183: struct ip_info conflict between esp_wifi.h and lwip/ip_addr.h

PieBru, 2016-08-13:
Hi, after following the README to install esp-open-rtos in my Arch Linux, to successfully build the blink example I had to comment out the whole struct ip_info definition into esp-open-rtos/include/espressif/esp_wifi.h

It conflicts with the same struct definition contained into esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/include/lwip/ip_addr.h
The function prototype sdk_wifi_set_ip_info contained into esp_wifi.h still gives a warning, I haven't verified if it's totally harmless, but at least blink builds successfully and the led blinks :-)

Warning detail:
../../include/espressif/esp_wifi.h:52:52: warning: 'struct ip_info' declared inside parameter list [enabled by default]
its scope is only this definition or declaration, which is probably not what you want [enabled by default]

Thanks,
Piero

sheinz, 2016-08-13:
Hi @PieBru,

For some reason your build is using *esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/include/lwip/ip_addr.h* but it should use our own **lwpi** source code.
In our lwpi/ip_addr.h the struct ip_info is conditionally compiled: [ip_addr.h](https://github.com/SuperHouse/esp-lwip/blob/3cf8d514bd76e6ef77e6fa514d0ec6d96da7fd9a/src/include/ipv4/lwip/ip_addr.h#L240)

projectgus, 2016-08-16:
We recommend building esp-open-sdk with STANDALONE=n for these kind of reasons. With STANDALONE=y (the default), esp-open-sdk copies headers/libraries for Espressif's SDK (and the esp-open-sdk LWIP) into the toolchain sysroot alongside the other "system" headers/libraries.

kanflo, 2016-08-25:
@PieBru, did you make any progress on this issue? Was it rested to STANDALONE=y?

PieBru, 2016-08-26:
Sorry for the delay, I should reach my pc the next week.
P.

PieBru, 2016-09-08:
I confirm @projectgus note: my issue is solved by building esp-open-sdk with STANDALONE=n
I installed and used esp-open-sdk before trying esp-open-rtos, so I didn't care about rebuilding it STANDALONE=n
Thank you.
Piero

closed 2016-09-08 by PieBru