SuperHouse/esp-open-rtos #184: making examples/posix_fs gives errors

yesco, 2016-08-14:
Wanted to try out the spiffs got into trouble trying to build the example, as test-fs was completely empty. The readme didn't mentioning anything so I thought it was missing files in the check-in.

Then checking around and googling I found https://travis-ci.org/SuperHouse/esp-open-rtos that showed the git submodules command:

git submodule update --init --recursive

Didn't find it mentioned anywhere on the setting up on the esp-open-rtos home page, or any message in the readme for the posix_fs.

Not sure what is the preferred way to make this easier.

Is there any drawbacks to always get all the submodules? I assume they don't take up space unless one is using the EXTRA_COMPONENTS?

sheinz, 2016-08-14:
Hi @yesco,

If you do a fresh clone with `git clone --recursive https://github.com/Superhouse/esp-open-rtos.git` as described in the README.md of esp-open-rtos it should take care of all the submodules in the project.

If you pulled changes and a new submodule appeared in the project it will not be pulled automatically. The following command will pull a new submodule: 
`git submodule update --init --recursive`

yesco, 2016-08-14:
Argh, I looked for it before sending a comment. Sorry, I just pulled and had troube, found the fix, but thought the document wasn't updated. My mistake.

closed 2016-08-14 by yesco