SuperHouse/esp-open-rtos #188: Silencing serial port

Ruphobia, 2016-08-15:
I've searched for hours on how to silence the uart without much luck.  No matter what I do, I still get unwanted output on uart0.  I removed all of the offending printf statements from the source and hex edited the binary blobs to remove the rest.  I finally have a working solution with a silent uart;  I have a few questions:

1) Did I miss something and is there another approach to keeping the uart quiet?  If so, can someone point to some source code example on how to do this?

2) Although a dirty approach, is hex editing the binary blobs an acceptable hack-around to this problem until the binary blobs can be replaced? and is it worth the time to create a patch that we can include with this project so others can enjoy a silent uart?

sheinz, 2016-08-15:
Hi @Ruphobia,

The new SDK from Espressif has function **system_set_os_print(0)** which silence SDK output to uart0. Unfortunately the SDK that eso-open-rtos is originally based on didn't have such a function. Also new versions of RTOS SDK from Espressif don't have such function.

I don't know any other way to silence libraries output.

I think the right approach would be to reverse engineer binary libraries and rewrite them in C as @foogod did with *open_esplib*. But it too much of work. Or look how **system_set_os_print** work in Espressif SDK and try to apply the same approach.

Ruphobia, 2016-08-15:
Excellent @sheinz,

This answers my first question and makes me feel all warm and fuzzy because  I didn't miss anything.

Now we just need to wait and see what others say about question #2 and if my method for hex editing the binary blobs is acceptable until we can reverse engineer the other libs...

projectgus, 2016-08-16:
I think there is a less invasive way to silence the SDK from printing. It's not very obvious though, so don't blame yourself for missing it!

In ld/program.ld there is a line:

```
ets_printf = printf;
```

... the SDK functions use Espressif's ets_printf, and this line maps that function to newlib's standard printf.

I'm fairly sure if you remove this line, and implement a function sdk_ets_printf() somewhere in your own code, then all print statements originating in the SDK should end up in your function. Make this function a no-op, and you should get the same result.

sheinz, 2016-08-16:
I tried @projectgus's suggestion. Unfortunately it didn't help.

I checked symbols that binary libraries use:
```
> xtensa-lx106-elf-nm libnet80211.a |grep print
xtensa-lx106-elf-nm: ieee80211_hwmp.o: no symbols
         U printf
         U printf
xtensa-lx106-elf-nm: ieee80211_mesh.o: no symbols
xtensa-lx106-elf-nm: ieee80211_node.o: no symbols
xtensa-lx106-elf-nm: if_eagle.o: no symbols
         U printf
         U ets_printf
         U printf
         U printf

```
So, the binary libraries use *printf* along with *ets_printf*. But it can be fixed :)
```
> xtensa-lx106-elf-objcopy --redefine-sym printf=ets_printf libnet80211.a
```
The same can be done with other libraries.
In the result all output from the binary libraries goes through function *ets_printf* which we can easily redefine and silence.

Ruphobia, 2016-08-16:
Excellent work  @sheinz @projectgus,

works great:

Step 1:

xtensa-lx106-elf-objcopy --redefine-sym printf=ets_printf libnet80211.a
xtensa-lx106-elf-objcopy --redefine-sym printf=ets_printf libwpa.a
xtensa-lx106-elf-objcopy --redefine-sym printf=ets_printf libpp.a

Step 2:
Edit esp-open-rtos/ld/program.ld and comment out "ets_printf = printf;"

Step 3:

Add dummy function:

int ets_printf(const char *format, ...)
{
	return 0;
}

Clean, recompile = silent uart.......

Ruphobia, 2016-08-16:
Now it would be interesting / possible to implement "system_set_os_print" function.  we could commit the modified libs to the project tree.  permanently remove the line in program.ld, then implement ets_printf somewhere, libmain maybe?

ets_printf default functionality would be to call printf.  if you call system_set_os_print(0), then it would just return.

sheinz, 2016-08-16:
@Ruphobia I was actually thinking the same. But I'm not sure if committing modified libraries is a good idea.

Ruphobia, 2016-08-16:
Is there a way to temporary rename during compile / link so we don't have to permanently modify?

projectgus, 2016-08-17:
Yes, there's a sequence of make targets that process the binary libraries:
https://github.com/SuperHouse/esp-open-rtos/blob/master/common.mk#L164

There is already a list of symbols that get renamed in lib/allsymbols.rename. If you add "printf ets_printf" here then everything should fall into place.

It might be nice to have a way for people to disable ets_printf without needing to modify the linker script, as well (not sure what the easiest approach for this is, short of just adding an intermediate function call).

(Regarding the libraries going through the "make" process: This is to prevent the need to commit binary modifications (ie the libs in our repo are the same ones Espressif shipped in the 0.9.9 RTOS SDK.) Originally this was because I anticipated that we'd keep updating the binary libraries as Espressif brought out new releases of the RTOS SDK, but when they switched from "MIT License" to "Espressif MIT" I cancelled this plan.)

projectgus, 2016-08-17:
> It might be nice to have a way for people to disable ets_printf without needing to modify the linker script, as well (not sure what the easiest approach for this is, short of just adding an intermediate function call).

I missed the suggestion that we just implement system_set_os_print() ourselves. This seems sensible.

Ruphobia, 2016-08-17:
Here's my changes that implement  system_set_os_print and silence the serial port.

That makefile makes my head hurt, but I gave it a go.

Let me know what you think...

[silentserialportdiff.txt](https://github.com/SuperHouse/esp-open-rtos/files/423730/silentserialportdiff.txt)

Ruphobia, 2016-08-19:
Found an interesting behavior of the compiler wile testing my patch in the previous comment.  I was getting a 0x0D 0x0A out of my serial port whenever I would connect to a station.  I tracked down the line of email code:  printf("\n"); in user_interface.c

the fix was to add the newline to the end of the printf in the previous line:

   printf("ip:%d.%d.%d.%d,mask:%d.%d.%d.%d,gw:%d.%d.%d.%d\n", ip_bytes[0], ip_bytes[1], ip_bytes[2], ip_bytes[3], mask_bytes[0], mask_bytes[1], mask_bytes[2], mask_bytes[3], gw_bytes[0], gw_bytes[1], gw_bytes[2], gw_bytes[3]);

I can only guess that the compiler was replacing the printf("\n") with a putc or something as it was bypassing my bypass :)

closed 2016-08-22 by Ruphobia

imbizos, 2017-10-27:
Maybe for someone will be helpful:

for lib in *.a ; do xtensa-lx106-elf-objcopy --redefine-sym printf=my_dummy_printf $lib; done
xtensa-lx106-elf-ar -xv libmain.a printf-stdarg.o
xtensa-lx106-elf-objcopy --redefine-sym my_dummy_printf=printf printf-stdarg.o
xtensa-lx106-elf-ar rcs libmain.a printf-stdarg.o