SuperHouse/esp-open-rtos #197: Firmware upgrade from host MCU

ole-johan, 2016-08-20:
I have an ESP module connected to a MCU. I would like to have this host MCU perform fw upgrade of the ESP. I don't want to use OTA upgrade, but rather have my host MCU manage this. Basically I would like a small implementation of a flashing tool like esptool.py, with only the flashing part, and in C. Before I start porting, does anyone have something like this, or a hint on some inspiration that might be useful?

Thanks.
Ole-Johan

kanflo, 2016-08-21:
I am not aware of any MCU side implementations but one way (if I am not mistaken) would be to connect to the SPI interface pins on the short side of the ESP12 and access the internal flash directly, while keeping the ESP in reset. Either way, I think you would find better help in the esp8266.com forums.

ole-johan, 2016-08-21:
Yes, SPI directly could be one way... In my current design I would prefer to keep the connection between the MCU and ESP simple, and only use the same uart as during normal operations. Now I have a simple mode in the MCU that boots the ESP from uart, and have a simple passthrough from the PC through the MCU to the ESP that uses esptool.py as if the MCU was not there. 
It would be cool to be able to do the exact same programming of the ESP from the MCU directly using the same transport. 
Thanks for the tips, I will bring this to the esp8266.com forum.

kanflo, 2016-08-24:
The MCU UART passthrough mode is a simple and clever trick. I will go ahead and close this issue if you don't mind.

closed 2016-08-24 by kanflo