SuperHouse/esp-open-rtos #200: sdk_system_deep_sleep_set_option in esp_system.h, but not in libmain.a

stephenmangan, 2016-08-25:
Hi, 

I've been testing the deep sleep options on the esp8266 using the sdk_system_deep_sleep() function. This works great, so I've moved onto looking at the power consumption when the sdk_system_deep_sleep_set_option() function is used to set the various wake up options, i.e. calibrate on start-up, do not calibrate, or leave modem off on start-up. 

Unfortunately, the calls to sdk_system_deep_sleep_set_option() are coming back with an undefined reference. I've searched libmain.a and there is no reference to (sdk_)system_deep_sleep_set_option in there, or in the allsymbols.rename file.

Is there another library that it should be called from?

Thanks,

Stephen

sheinz, 2016-08-25:
Hi Stephen,

It seems like there's no implementation of *system_deep_sleep_set_option* in  esp-open-rtos.

The binary libraries that esp-open-rtos is based on doesn't contain this function. It looks like the definition of the function was added into the header by mistake in https://github.com/SuperHouse/esp-open-rtos/commit/b18f2947ee4e854d915e619216d28e984ab63d37

stephenmangan, 2016-08-25:
Hi Sheinz,

Thanks for the quick response. 

I've had a quick look through the libmain library in the upstream espressif RTOS SDK v1.4, and seen that it contains the symbol for sdk_system_deep_sleep_set_option(). Are there any plans to update the esp-open-sdk in the near future?

Thanks,

Stephen

projectgus, 2016-08-26:
Hi Stephen,

> Are there any plans to update the esp-open-sdk in the near future?

There are no current plans. When I started the project the Espressif SDK version was 0.9.9 and it was licensed MIT. At that point I assumed we'd be upgrading to future version. However. for the next SDK release the RTOS SDK changed licenses from MIT to Espressif MIT. This precludes relicensing parts of it as BSD or another OSI-approved open source license. For this reason there hasn't been an SDK update, see https://github.com/SuperHouse/esp-open-rtos/#contributions for another view of this.

If you want to discuss licensing further the best place to do it is probably the mailing list: https://groups.google.com/forum/#!forum/esp-open-rtos 

However, regarding `system_deep_sleep_set_option` - what this function does is sets a byte in the RTC memory which controls the behaviour on coming out of deep sleep (due to RTC memory persisting after the deep sleep reset.) On reset the SDK startup code checks this byte and behaves accordingly. Because we have open source startup code, it is possible to add whatever deep sleep control functionality is needed into esp-open-rtos without relying on the binary SDK to do this - we just need to choose some RTC memory to use for this function, and work out what control logic we need in the startup code.

I suggest we do this by creating a new API, because AFAIK system_deep_sleep_set_option() is not fully understood and we can probably improve on it with something better documented.

stephenmangan, 2016-09-06:
Hi Projectgus,

Thanks for the explanation, and thanks for all the hard work in developing the esp-open-rtos sdk. 
I'll take a look at the start-up code and how it behaves during start-up.

Thanks,

Stephen

closed 2016-09-06 by stephenmangan