SuperHouse/esp-open-rtos #201: Refactor symbol names in MQTTESP8266

resetnow, 2016-08-25:
Many function names in MQTT client port don't have a common prefix (NewNetwork, ConnectNetwork, Network, expired, left_ms), and they all reside in a global namespace.This can lead to confusion, and overshadowing by local variables and functions. If it's okay with maintainers, I can open a PR with these symbols renamed to either camelcase format (MQTTNetwork, MQTTNetworkNew, MQTTNetworkConnect) or “snake case” (mqtt_network_t, mqtt_network_new, mqtt_network_connect). I suggest the former because Paho MQTT uses it.

kanflo, 2016-08-25:
I think that is a good idea. Please sync with @baoshi who did the MQTT port.

resetnow, 2016-08-25:
Could you please elaborate on syncing? Should I ask @baoshi to do the renaming? Or create a PR against his MQTT repo and port the changes here afterwards?

projectgus, 2016-08-26:
FWIW, I agree on both counts, but I think some of this also goes back to the Eclipse Paho MQTT C port which provides names for some of the functions. So I think we need to change both esp port-specific names and the names used by that project.

@baoshi may have some ideas on the best way to go forward with this. I guess just renaming everything and moving away from the Paho MQTT names is the best option for clean namespacing.

resetnow, 2016-08-26:
>I guess just renaming everything and moving away from the Paho MQTT names is the best option for clean namespacing

I guess we can have “snake case” names then, because e-o-r core API uses it.

resetnow, 2016-08-26:
#204

closed 2016-09-18 by resetnow