SuperHouse/esp-open-rtos #206: add comment in i2s audio sample on build instructions

phlemoine, 2016-08-26:
suggesting to add 
 * Build instructions: cd esp-open-rtos; make -j4 -C examples/i2s_audio
 * Build and flash/test : cd esp-open-rtos; make flash -j4 -C examples/i2s_audio ESPPORT=/dev/ttyUSB0
 *
in examples/i2s_audio/i2s_audio_example.c on line 18

(i just struggled to build and flash this example properly, so hoping to help future users:-) )

sheinz, 2016-08-27:
Hi @phlemoine,

Building *i2s_audio* example is no different from any other example. The instructions are on the main README.md:

> Build an example project (found in the 'examples' directory) and flash it to a serial port:
make flash -j4 -C examples/http_get ESPPORT=/dev/ttyUSB0
Run make help -C examples/http_get for a summary of other Make targets.

>(Note: the -C option to make is the same as changing to that directory, then running make.)

kanflo, 2016-08-27:
Welcome to the project @phlemoine. I agree with @sheinz that individual examples need no included build instructions unless there is something exotic going on but I do appreciate your willing to help future users.

phlemoine, 2016-08-27:
thanks for the feedback. i had no issues building my first example, but forgot about the way it works when i built the second one. it took me quite some times and web search to remember about calling make from the "main" directory and not from the example directory itself. maybe a generic ref to the instructions would make it.

sheinz, 2016-08-28:
>it took me quite some times and web search to remember about calling make from the "main" directory and not from the example directory itself.

It should work from the example directory as well:
```
cd examples/i2s_audio/
make
```

phlemoine, 2016-08-28:
indeed it does. not sure what i was doing wrong. i am closing the issue

closed 2016-08-28 by phlemoine