SuperHouse/esp-open-rtos #225: program too large with commit 12d0d

phlemoine, 2016-09-19:
i just rebased on head of the repo (currently at 83689) and found out my program no longer fits the max size due to commit 12d0d. has anyone any recommendation on how to adjust this commit or a local change i could use to overcome this issue and still be able to stay aligned with the latest and greatest version?

resetnow, 2016-09-30:
Have you tried adding `-Os` to CFLAGS?

phlemoine, 2016-09-30:
i just dit, with this :
CFLAGS:=$(filter-out -O2,$(CFLAGS))
CFLAGS += -Os

however, the error is still reported:
section `.text' will not fit in region `iram1_0_seg'

resetnow, 2016-10-01:
Most functions should be placed into flash by default, do you explicitly specify any function attributes in your code to force functions into IRAM?

phlemoine, 2016-10-01:
Nope. All functions run in flash. What puzzles me is what in this commit
(12d0d) is causing this error. Once reverted, it builds fine

resetnow, 2016-10-02:
This commit increases data memory usage by a few bytes, maybe that's the reason? Try building any of example apps (with `Os`), and maybe `make clean` in your app to rebuild everything with optimization.

resetnow, 2017-01-13:
Out of curiosity, did you figure it out?

phlemoine, 2017-01-14:
No, I didn't. I went back to the code changes on that commit, and can't
really spot what would cause such an error.
I had try to build the examples , but that didn't help me with the issue.
I'll conduct some more experiment next week and will post the results.

>