SuperHouse/esp-open-rtos #226: reduce execution time of mbedtls_ssl_handshake()

chennakeshavaSoC, 2016-09-20:
Hi,

I am using esp-open-rtos sdk integrated with AWS IoT, everything is perfect as a first step.

When looking at the timing stats, mbedtls_ssl_handshake() is taking around 15+ seconds to return, connected successfully to aws cloud. Would it be possible to reduce this time, so that I can respond quickly, as my device is battery powered sleepy node, acts on events.

phlemoine, 2016-09-27:
have you tried using overcloking during the initialization phase ? looks like it goes down to ˜7 secs

#include <esplibs/libmain.h>                                                                                                                                    sdk_system_overclock();
[...]
sdk_system_restoreclock();

resetnow, 2016-09-27:
I wonder if there is a crypto core in ESP8266 like in ESP32

chennakeshavaSoC, 2016-10-05:
No luck with the overclocking, it is still taking same 15 seconds for ssl connect.

recursify, 2017-06-17:
@chennakeshavaSoC I know this is a bit late, but I was digging into some similar issues.  If you're using ECDH, it looks like mbedtls prioritizes some pretty slow elliptic curves, most notably `secp521r1` and the `brainpool` ones.  `secp256r1` looks to be the default in the latest versions of OpenSSL so is probably a reasonable default.  It's worth inspecting the TLS handshake using wireshark to see what curves are being offered by the ESP, and selected by AWS IoT.  

To disable some of the slower curves:

```
#undef MBEDTLS_ECP_DP_SECP521R1_ENABLED
#undef MBEDTLS_ECP_DP_BP512R1_ENABLED
#undef MBEDTLS_ECP_DP_BP384R1_ENABLED
#undef MBEDTLS_ECP_DP_BP256R1_ENABLED
```

You can also use `mbedtls_ssl_conf_curves` to reorder the curves.

closed 2022-03-23 by chennakeshavaSoC