SuperHouse/esp-open-rtos #227: xNotifyTask APIs

phlemoine, 2016-09-20:
i am looking at using xTaskNotify APIs (http://www.freertos.org/RTOS-task-notification-API.html) but can't  get any hits with grep (and compilation fails too obviously). have these APis been excluded from the port ? how easy is this to add this from freertos code base ?

projectgus, 2016-09-20:
Hi @phlemoine,

The Task Notificiation APIs were now in FreeRTOS 8. esp-open-rtos is still on version 7.

There is a PR to upgrade to FreeRTOS 9 open in #169. At last look, it needed some testing and checks before merging to mainline - if you can help with any testing of this it would be great.

Angus

sheinz, 2016-10-26:
FreeRTOS is upgraded with https://github.com/SuperHouse/esp-open-rtos/pull/245

closed 2016-10-26 by sheinz