SuperHouse/esp-open-rtos #229: ESP exception occuring first time on AP mode after power cycle

chennakeshavaSoC, 2016-09-30:
Hi,
We are using esp-open-rtos with ESP-12F in AP mode, we are seeing ESP exception first time on every power cycle, as soon as a device is joined to this AP.

Is there any solution for this?

kanflo, 2016-10-05:
Normally it should not. Are you using the access_point example program? If not, could you try it?

projectgus, 2016-10-07:
When you say "on every power cycle", does it come back from the exception reset and work afterwards? Very odd!

Do you have some log output that includes the exception trace? (You can run it through utils/filteroutput.py to resolve symbol names from addresses.)

sheinz, 2016-11-02:
Hi @chennakeshavaSoC, it's been a while since you reported the problem. Have you solved it? Can we close the issue?

closed 2016-11-25 by sheinz