SuperHouse/esp-open-rtos #230: Patch: QEMU simulation of ESP8266 prior to flashing

dgtlrift, 2016-10-07:
[qemu.patch.txt](https://github.com/SuperHouse/esp-open-rtos/files/516615/qemu.patch.txt)
This patch adds support for running a project through QEMU for simulation prior to loading to flash.

`make simulate -C examples/http_get`

then to obtain simulated serial port output
`telnet localhost 4444`

Patch relies on [QEMU-xtensa with ESP8266 machine support](https://github.com/OSLL/qemu-xtensa/commits/xtensa-esp8266) being installed.  I suppose QEMU-xtensa could be a sub-module - but it would probably make better sense to be a submodule of esp-open-sdk and just referenced here as part of the sdk.

sheinz, 2016-10-21:
I tried it and it does work.
But it seems like qemu-esp8266 is missing SPI flash support. Sysparam which is now part of the eor core is failing at start.

AndreaGreco, 2016-10-28:
Hello all,

I try with this step:
[http://wiki.linux-xtensa.org/index.php/](http://wiki.linux-xtensa.org/index.php/Xtensa_on_QEMU#Setting_up_and_running_Xtensa_simulation_on_QEMU)
I follow this  step, and it arrive at and and create quemu, but it does not provide support for ESP8266, `qemu-system-xtensa -M help` does not figure ESP8266.

I found that there was a patch in a brank over this repo:
[OSLL/qemu-xtensa/commits/xtensa-esp8266](https://github.com/OSLL/qemu-xtensa/commits/xtensa-esp8266)
It compile but produce lot's of output over qemu teminal this:
```
esp8266_spi_write: +0x00 = 0x08000000
esp8266_spi_read: +0x00: 0x00000000
esp8266_spi_read: +0x10: 0x00000002
esp8266_spi_write: +0x04 = 0x201fe1c0
esp8266_spi_write: +0x40 = 0x00000000
esp8266_spi_write: +0x44 = 0x00000000
esp8266_spi_write: +0x48 = 0x00000000
esp8266_spi_write: +0x4c = 0x00000000
esp8266_spi_write: +0x50 = 0x00000000
esp8266_spi_write: +0x54 = 0x00000000
esp8266_spi_write: +0x58 = 0x00000000
esp8266_spi_write: +0x5c = 0x00000000
esp8266_dport_read: +0x0c: 0x04000002
esp8266_spi_write: +0x10 = 0x00000000
esp8266_spi_write: +0x00 = 0x08000000
esp8266_spi_read: +0x00: 0x00000000
esp8266_spi_read: +0x10: 0x00000002
esp8266_spi_write: +0x00 = 0x40000000
esp8266_spi_read: +0x00: 0x00000000
esp8266_spi_write: +0x10 = 0x00000000
esp8266_spi_write: +0x00 = 0x08000000
esp8266_spi_read: +0x00: 0x00000000
```

I try to compile this repo with brank "xtensa-fixes":
[mmaret/qemu-xtensa/tree/xtensa-fixes](https://github.com/mmaret/qemu-xtensa/tree/xtensa-fixes)

I compiled it with this configure-script:
`./qemu/configure --prefix=`pwd`/root --target-list=xtensa-softmmu`

It compile but not work, no messages in telnelt terminal tab, no messages in qemu terminal tab.
But telnet open connection then qemu is here to reply to sync, send from telnet.
Someone know compilation step/ configuration or repo where take patch.

My gratitude who develop:qemu.patch.txt working well good patch.

----------
Andrea

dgtlrift, 2016-10-29:
The following worked for me.. but it doesn't look too different from yours.  I have't tried with the branch you've referenced in your post.
```
        ../qemu/configure                       \
                --prefix=/usr/local  \
                --target-list=xtensa-softmmu    \
                --disable-werror        &&      \
        make                            &&      \
        sudo make install

```

AndreaGreco, 2016-10-29:
Tanks for your reply,

I write here comple procedure
```
git clone --recursive git@github.com:OSLL/qemu-xtensa.git
git checkout xtensa-esp8266
../qemu/configure                       \
                --prefix=/usr/local  \
                --target-list=xtensa-softmmu    \
                --disable-werror        &&      \
        make                            &&      \
        sudo make install
```
It work on me, with sample blinky or ds18b20_onewire.

There some problem with example that use networking function like http_get.
I think that emulator does not support networking functions.

Andrea

dgtlrift, 2016-11-04:
I've started peeling back the interface that @jcmvbkbc has implemented for the ESP8266... It will take me a bit of a task, but I believe you're correct thinking that the network isn't completely supported.  I'm looking to see what and how to emulate the radio for QEMU.

In any case, I believe having this mechanism can assist with quick prototyping with esp-open-rtos, and also vet and expose the gaps in the ESP8266 QEMU interfaces.

AndreaGreco, 2016-11-05:
Stay in touch if you do some progress over this.