SuperHouse/esp-open-rtos #231: esp_spi.c

only1chi, 2016-10-07:
I've been trying to get SPI port 1 going on my esp8266 (NodeMCU) board. 
I am using esp_spi.c driver from the esp-open-rtos core library.
I configured the pins for SPI port 1 per the pin out in esp_spi.c. The pins on the SPI port do not transition. I don't see the SCLK moving on the scope. 
Has anyone used, or is anyone using the esp_spi.c file for SPI transactions on SPI1 port?

UncleRus, 2016-10-08:
I'm using SPI1 with NodeMCU and MAX7456.
NodeMCU pinout:
[image]

You can test it with minimal program:
```
#include "espressif/esp_common.h"
#include "esp/uart.h"
#include "esp/spi.h"

void user_init(void)
{
    uart_set_baud(0, 115200);
    printf("SDK version:%s\n", sdk_system_get_sdk_version());

    const spi_settings_t my_settings = {
	 .mode = SPI_MODE0,
	 .freq_divider = SPI_FREQ_DIV_4M,
	 .msb = true,
	 .endianness = SPI_LITTLE_ENDIAN,
	 .minimal_pins = true
	};
	const char out_data [] = "HELLO WORLD";
	char in_data [sizeof(out_data)];

	spi_set_settings(1, &my_settings);
	while (true)
	{
		spi_transfer(1, out_data, in_data, sizeof(out_data), SPI_8BIT);
		in_data[sizeof(in_data) - 1] = 0;
		printf("Sent: %s, got: %s (", out_data, in_data);
		for (size_t i = 0; i < sizeof(in_data); i ++)
			printf("0x%02x ", in_data[i]);
		printf(")\n");
	}
}
```

only1chi, 2016-10-08:
When I run this test, should I still use "esp_spi.c" since the functions you use:
spi_set_settings
spi_transfer

are all defined in esp_spi.c

I did run the test with esp_spi.c included in my project, and I get all "0x00".
What is sent is NOT what is received. 

I'm running the test with the module stand alone, nothing else connected to it. Since this is a wraparound test it shouldn't need to be connected to any other device.

UncleRus, 2016-10-08:
Try to short GPIO12 and GPIO13. In terminal you should see something like this:
```
Sent: HELLO WORLD, got: HELLO WORLD (0x48 0x45 0x4c 0x4c 0x4f 0x20 0x57 0x4f 0x52 0x4c 0x44 0x00 )
```

only1chi, 2016-10-09:
@UncleRus I did short GPIO12 and GPIO13 (MISO and MOSI), but I didn't get the same output you're getting. 
I'm getting all 0x00 from the the HSPI port.

only1chi, 2016-10-09:
Please note that I did compile file esp_spi.c with my project.

only1chi, 2016-10-09:
Where can one find the actual detailed reference guide for the esp8266 processor? 
I haven't seen any reference guide that has details on register mappings and how each module works.

only1chi, 2016-10-09:
@UncleRus ... I read your previous comments and compiled the simple example with the code you provided. It is working!

I had taken the esp_spi.c file in core.c and built it as part of my project. I think I confused the linker since, the files in the esp-open-rtos core folder are actually built as a library. The functions I was calling weren't doing anything because they weren't liking to the "core" library.

I will remove esp_spi.c from my project.

Thanks for your help.

closed 2016-11-02 by sheinz

DitroniX, 2020-06-06:
This is an old thread but wonder if anyone has got code working on an ESP32 and MAX7456?