SuperHouse/esp-open-rtos #233: No example working (http_get & http_get_mbedtls)

ameeuw, 2016-10-10:
Hello, 
I am trying to get the esp-open-rtos SDK to work as I need the mbedTLS crypto framework. Until now I have been successful with setting up the toolchain and compiling the examples is working fine (no errors after following the instructions in the projects readme).
I have flashed the resulting firmware image to my esp8266 boards (WeMos D1 mini with ESP12e module) but the software only spits out garbage at any given baud-rate.

My question is: Am I missing something crucial in getting the examples to work? I have replaced the WiFi settings.

I am thankful for any hints what I could do differently.
Best,
Arne

kapitanvong, 2016-10-10:
What are you using for serial port reading?

ameeuw, 2016-10-10:
I would like serial output to see if it works and to go on developing.

The example (http_get_mbedtls) should output at start:
printf("HTTP get task starting...\n");
printf("\n  . Seeding the random number generator...");

and whether it failed or succeeded...

kapitanvong, 2016-10-11:
Try moserial if you are on Linux or Esplorer if you are Windows user.

ameeuw, 2016-10-11:
I'm using coolterm and screen. My previously used firmware is nodemcu and everything is working fine with that one.
Maybe it is something with the initialisation of the firmware? Or could something be wrong in the build process?
Thank you for bearing with me here.

kapitanvong, 2016-10-11:
What line do you use to build and flash project? Does "make flash -j4 -C examples/http_get ESPPORT=/dev/ttyUSB0" works fine for you?  You mentioned you followed the instructions for building toolchain and compiling, but I don't no if you do flashing and compiling separately.
You can try to listen on baud rate 74880, which should get you some boot info in very beginning.

ameeuw, 2016-10-11:
I am flashing separately as I am running the build environment in a virtual machine. When I listen in at boot time at 74880 I get the following message:

```
ets Jan  8 2013,rst cause:2, boot mode:(3,7)

ets_main.c 
```

ameeuw, 2016-10-11:
I also tried building an example using bschwind's docker image [dockerhub link](https://hub.docker.com/r/bschwind/esp-open-rtos/) and get the same result.

Am I crucially missing something in the flashing process? I never had problems with flashing firmware images (micropython, nodemcu etc.) before so I am wondering where I am going wrong...

@kapitanvong What is the '-j4' flag in the building process? Am I missing something in terms of building for flash size?

resetnow, 2016-10-11:
1. Have you tried setting UART speed manually in code?
1. esp-open-rtos writes several flash regions at start:
   
    ```
    esptool.py --port /dev/ttyUSB0 write_flash \
    	--flash_freq 40m \
    	0x0 ./esp-open-rtos/bootloader/firmware_prebuilt/rboot.bin \
    	0x1000 ./esp-open-rtos/bootloader/firmware_prebuilt/blank_config.bin \
    	0x02000 your-firmware-0x02000.bin
    ```
    
    If you flash using some other script — maybe you need to erase configuration memory region (blank_config).

ameeuw, 2016-10-11:
Thank you so much, I did not look into the flashing procedure. I adapted my flashing script and now everything works. Thank you so much.

closed 2016-10-11 by ameeuw