SuperHouse/esp-open-rtos #234: Inability to sniff ESP8266 packets using Wireshark

kapitanvong, 2016-10-10:
I use example code from both sntp and http_get_mbedtls, and I get valid output through serial showing ntp time each 5 secs and https://www.howsmyssl.com/ info each 10. But while capturing packets with Wireshark, only info that I could get is a ICMP response messages, when I try to ping device in the process. I tried filtering using IP, MAC by values provided in serial on the start-up with no success. Same with NTP and TLS. Such a mystery.

dgtlrift, 2016-10-10:
Were you sniffing from the machine you were initiating the ping from? Was the WiFi interface in monitor mode - analogous to Ethernet promiscuous mode.

kapitanvong, 2016-10-11:
I see mine wifi adapter (wlp2s0) as ethernet here, and I have promiscuous mode tick on. I can see everything on my local besides packets from ESP8266. I could capture them when I was using latest espressif RTOS SDK 1.4.0 and raw TCP and UDP sockets.
[image]

projectgus, 2016-11-10:
If your WiFi network uses WPA then wireless clients are isolated at the network level, so promiscuous mode by itself doesn't let you see other clients' traffic.

For this kind of debugging I have two approaches:
* Run openwrt on the WiFi AP, and run tcpdump there then copy the files to my local computer to load in Wireshark. (It's all possible to run a "mirror port" but I haven't gotten around to trying this.)
* I have a second USB WiFi adapter that I set up as a tiny access point (with hostapd) and then route traffic onto my main network (using iptables) so my laptop becomes a small router. Then the ESP8266 connects to that tiny access point.