SuperHouse/esp-open-rtos #236: uart 1 not working

ptmeco, 2016-10-11:
I have a problem getting uart 1 to work. Is far as I understand  the code below should:
- set speed for both serials,
- switch printf output to uart 1, 
- print SDK version on uart 1,
- print 10 'X' on uart 0 and 10 'b' on uart 1.

When I ran that code uart 1 is completely quiet.  `sdk_os_install_putc1` has no effect, printf is still on uart 0.
10 'X' appears on uart 0 (which is ok). 
What am I doing wrong ?

````
  #include "espressif/esp_common.h"`
  #include "esp/uart.h"
  #include "esp/spi.h"
  void myPutc(char c){
      uart_putc(1,c);
  }
  void user_init(void){
      uart_set_baud(0, 115200);
      uart_set_baud(1, 115200);
      sdk_os_install_putc1(myPutc);
      printf("SDK version:%s\n", sdk_system_get_sdk_version());
      for (int a =0 ;a <10;a++) {
        uart_putc(0,'X');
        uart_putc(1,'b');
     }
  }
````

sheinz, 2016-10-12:
To get output from TX of UART1 you need to select "UART TX" function on this pin
```
gpio_set_iomux_function(2, IOMUX_GPIO2_FUNC_UART1_TXD);
```
As far as I understand you can't get **printf** to output to **UART1** as it is hard-coded [here](https://github.com/SuperHouse/esp-open-rtos/blob/master/core/newlib_syscalls.c#L55)

sheinz, 2016-10-12:
But if you really need to get **printf** to output to UART1 you can locally change the uart number in [newlib_syscalls.c](https://github.com/SuperHouse/esp-open-rtos/blob/master/core/newlib_syscalls.c#L55) 
Or propose a feature to select what UART port is used for stdin/stdout.

(Thanks for the great job bringing this part of the framework open source!)

closed 2016-11-25 by sheinz