SuperHouse/esp-open-rtos #238: code uploaded succeeds but ESP8266 doesn't run (NodeMCU) for ESP8266_RTOS_SDK v1.4.0

sarwadenj, 2016-10-14:
Hi,
I'm trying to run blinking LED program using ESP8266_RTOS_SDK.
program link-->https://github.com/sarwadenj/ESP8266_RTOS/tree/master/test_2
also i written another program and set WiFi in station mode--->https://github.com/sarwadenj/ESP8266_RTOS/blob/master/wifi/user/user_main.c

I compiled my program using 
./gen_misc.sh
and set 
STEP 1:    use boot_v1.2+ by default
STEP 2:    generate bin: user1.bin
STEP 3:    spi speed: 40 MHz
STEP 4:    spi mode: QIO
STEP 5:    spi size: 4096KB
                spi ota map:  1024KB + 1024KB

and uploaded code going into Bin_path-->
sudo esptool.py --port /dev/ttyUSB0 write_flash --flash_mode qio 0x2000 user1.4096.new.6.bin

it uploaded successfully but code didn't run. So what should be the problem. is I'm missing something.

Is there any solution for this?
Thanks in advance.

sheinz, 2016-10-14:
It seems like you are not using esp-open-rtos at all.

If you use esp-open-rtos you don't need any other SDK (like ESP8266_RTOS_SDK v1.4.0)

Also esp-open-rtos has it's own build system. You don't need to run any scripts (like ./gen_misc.sh) to build a project. You simple run `make` and `make flash`

sarwadenj, 2016-10-15:
Hi @sheinz how i setup it in lubuntu.
for that i have to install esp-open-rtos or esp-open-sdk.

resetnow, 2016-10-15:
@sarwadenj both, see https://github.com/SuperHouse/esp-open-rtos#quick-start

sarwadenj, 2016-10-17:
@resetnow so esp-open-sdk i have to install in which mode 
make STANDALONE=y
or
make toolchain esptool libhal STANDALONE=n.

resetnow, 2016-10-17:
@sarwadenj if you already ran make then run `make clean` and then `make toolchain esptool libhal STANDALONE=n`.

sarwadenj, 2016-10-19:
@resetnow as esp-open-rtos is based on  Espressif RTOS SDK 0.9.9 so where i will get API provided espressif of Espressif RTOS SDK 0.9.9.

sarwadenj, 2016-10-20:
@sheinz ,@resetnow
Hi,

Previously i am programming ESP8266 in Arduino IDE. 
I want to use ESP8266.CPP and ESP8266.h library for my RTOS based code in C. 
So how can i use C++ CODE in my ESP-OPEN-RTOS based C CODE.

Thanks.

closed 2016-11-02 by sheinz