SuperHouse/esp-open-rtos #240: Crash - Fatal Exception 29

toashwin, 2016-10-19:
I am seeing a crash when I run.

I was using a branch from: 
https://github.com/rongsaws/esp-open-rtos/

I see using following features: 
1) Serial read.
2) MQTT
2) AWS_IOT example updated with my code for serial read.

I am going to try the latest code. But for now, below is report. Any idea what is going on?

Below is the crash report. 
Fatal exception (29): 
epc1=0x40208dd0
epc2=0x00000000
epc3=0x40227b3c
excvaddr=0x00000000
depc=0x00000000
excsave1=0x40208dd0
Registers:
a0 40208dd0 a1 3fff5760 a2  3fffc820 a3  00000000
a4  3fff3798 a5  3fffc829 a6  3fffc828 a7  40229d30
a8  00000020 a9  00000020 a10 3fff155c a11 0000000a
a12 3fffc820 a13 00000021 SAR 0000001e

Stack: SP=0x3fff5760
0x3fff5760: 00ff0000 ff000000 3fff1514 40229b48
0x3fff5770: 3fff58f0 3fff58f4 00000021 4020903f
0x3fff5780: 65636572 64657669 74616420 30203a61
0x3fff5790: 61362032 20353020 66203862 37642065
0x3fff57a0: 00000020 40229a95 0000004e 60000000
0x3fff57b0: 0000000d 3fff155c 0000000a 3ffe800c
0x3fff57c0: 00000000 40229a93 3fff1514 0000001d
0x3fff57d0: 00000000 ffffffff ffffffff 00000000

Free Heap: 1832
_heap_start 0x3fff1470 brk 0x3ffff9b8 supervisor sp 0x3ffffae0 sp-brk 296 bytes
arena (total_size) 58696 fordblks (free_size) 1536 uordblocks (used_size) 57160
@[03] *¬Ô[15][02]ñ*%Dþ/‘"D°ƒ[12][02]ÈÄÕ(õ[0B] –*D°c$ü[08]üÕPê[0F][12]´b‚^–[12][08]-I¦ÓµýdEU€!
¶Ñ*[00]!+T[01]„[0C]^Q[0B][16]¶ƒ^Q[0B][02]5ñ^[17][1F]Š”ŸV[1F]Ì@´-§Ä„œ×Ä°ŽÞ(‰€<å[18][06][02]Ñ+!%[00]Þ×[16]cV$ŽEÍ•ž¿åŽy1gÑ+[18]©[00]Þ¦¨@ŽÎ[08]ÉŒHm§²ëDm[1C]d„ƒÓ˜Gvc V[0E]bÄ„ƒSrcI[08]Þàc[00]J
[18]èæ!N¤ÌSDK version: 0.9.9, free heap 43884
mode : sta(18:fe:34:d1:8f:e2)
add if0
wifi_task: Connecting to WiFi
wifi_task: status = 1
scandone
wifi_task: status = 1
add 0
aid 9
cnt

phlemoine, 2016-10-19:
You seem to have very low free heap (1832) . Also, check the
TaskHighWaterMark , as it may be underestimated, causing an overflow

toashwin, 2016-10-19:
OK.. 
How do I check/debug/fix on the heap? and TaskHighWaterMark 
Is there any documentation/pointers?

My code runs fine for few minutes and then crashes. Boots up again runs fine for few more minutes and crashes again.. 

thanks
Ashwin

phlemoine, 2016-10-19:
Are you successfully connecting to aws servers, or is the crash happening
before the connection goes thru?

You may want to not start some tasks until the service is connected so that
the TLS stack can use the required memory.

Add tracing points in the code with the following API:
uxTaskGetStackHighWaterMark
xPortGetFreeHeapSize
That way you can better understand what's happening and adjust the
code/flow accordingly.

toashwin, 2016-10-19:
Yes, I am able to connect to the AWS and post the data to the cloud successfully. But after 2min it crashes. Then restarts and connects back to AWS and correctly posts the data..   And this keeps happening every 2min,

Thanks
Ashwin

resetnow, 2016-10-19:
By your description you may have a memory leak somewhere.

Wazalogic, 2016-10-26:
I experience a similar problem when the program gets low on memory. If I turn off mqtt then all is ok. It is as if mqtt uses memory which the program can overwrite or otherwise. Could be a memory leak.

Are you using statics? The compiler/linker seems to have issues with statics in C++. This can raise this fatal exception.

toashwin, 2016-10-26:
It was memory leak.. Problem with my code. I fixed it.. Now let my device run continuously for 2days and it did not crash.

sheinz, 2016-10-26:
@toashwin I'm glad you've found the problem. I'll close the issue.

closed 2016-10-26 by sheinz