SuperHouse/esp-open-rtos #243: Use c wrapping of c++ code (use  C++ CODE in my ESP-OPEN-RTOS based C CODE.)

sarwadenj, 2016-10-20:
Hi,

Previously i am programming ESP8266 in Arduino IDE. 
I want to use ESP8266.CPP and ESP8266.h library for my  RTOS based code in C. 
So how can i use  C++ CODE in my ESP-OPEN-RTOS based C CODE.

Thanks.

sheinz, 2016-10-20:
Hi @sarwadenj,

You can't use classes and objects in c code. To use functions from c++ code in c you need to declare them with ` extern "C"`

[Have a look at recommendations on how to mix C and C++](https://isocpp.org/wiki/faq/mixing-c-and-cpp#call-cpp)

sarwadenj, 2016-10-20:
Hi @sheinz,
i want to connect MPU6050 to esp8266 and I've "MPU6050_6Axis_MotionApps20.h" header file. 
and it contains mpu6050 class and functions for that class like---->

bool MPU6050::dmpPacketAvailable() {
    return getFIFOCount() >= dmpGetFIFOPacketSize();
}

so how can i use it.

Thanks.

sheinz, 2016-10-21:
Hi @sarwadenj,

If your c++ code doesn't use c++ constructions (like classes and objects) than you can just place that code into .c file and it will be compiled as c source code.
But you can compile c++ code with *esp-open-rtos*, have a look at the [example](https://github.com/SuperHouse/esp-open-rtos/blob/master/examples/cpp_01_tasks/cpp_tasks.cpp)
And you don't need to additionally install c++ compiler.

sarwadenj, 2016-10-24:
Hi @sheinz ,
actually for my program i required c++ constructions (like classes and objects) and for that purpose i want to install c++ compiler. so how i install c++ compiler in esp-open-rtos.

sheinz, 2016-10-24:
Hi @sarwadenj,

If you can successfully build `examples/cpp_01_tasks` with `make -C examples/cpp_01_tasks` you already have everything you need.
You do not need to additionally install C++ compiler.

closed 2016-11-02 by sheinz