SuperHouse/esp-open-rtos #248: conflicting '_read_r'

SaimenSays, 2016-10-22:
Hello,

when using components "extras/spiffs" and "extras/stdin_uart_interrupt", compiler brings up an error because the function `_read_r()` is implemented twice. 
I think these `_r` functions are for posix support of spiffs. But there should be a compiler flag to enable this, because it can cause also some other sideefects.

ourairquality, 2016-10-23:
A hack workaround would be to rename the definition in stdin_uart_interrupt and call this from the definition in spiffs if it is not a spiffs file descriptor, replacing the current uart rx handling code in spiffs.

Perhaps a more permanent fix would be to integrate stdin_uart_interrupt into the core, perhaps behind a build option, and then conditionalize the spiffs code on this option. The stdin_uart_interrupt code seems like a core function to me.

SaimenSays, 2016-10-23:
@ourairquality: I don't need a hack, because I'm not using this posix read functions on spiffs.

The switch between outputs should be in core, but the functionalities configurable/overwritable...

sheinz, 2016-10-23:
Hi @SaimenSays,

Thanks for bringing this issue. I've created a [fix](https://github.com/SuperHouse/esp-open-rtos/pull/249). You can try it.

SaimenSays, 2016-10-23:
@sheinz: Thanks for this hint. I tested it right now with files from #249, but it does not work as expeccted! My call on `read(0, (void*)&byte, 1)` blocks all lower priority threads. I did no investigation, but I think something might be wrong.

Rethinking about the blocking call, I seriously want to have the interrupt driven uart as core functionality. The Idle task gets not executed anymore when calling the blocking read, so RTOS does not work properly anymore.

sheinz, 2016-10-30:
Hi @SaimenSays, can you try the updated https://github.com/SuperHouse/esp-open-rtos/pull/249
It should work as expected now.

closed 2016-11-01 by SaimenSays