SuperHouse/esp-open-rtos #251: trying to build https://github.com/baoshi/ESP-I2C-OLED

Walter1604, 2016-10-24:
Need help and advice

after trying several of the examples in ESP8266_RTOS_SDK, I tried to build ESP-I2C-OLED only to find that I get the following errors:
LD build/ESP-I2C-OLED.out
./build/core.a(app_main.o): In function dump_flash_config_sectors': /home/walter/esp-open-rtos/core/app_main.c:450: undefined reference touser_init'
./build/core.a(app_main.o): In function sdk_user_init_task': /home/walter/esp-open-rtos/core/app_main.c:329: undefined reference touser_init'
collect2: error: ld returned 1 exit status
../../common.mk:198: recipe for target 'build/ESP-I2C-OLED.out' failed
make: *** [build/ESP-I2C-OLED.out] Error 1

walter@walter-VB:~/esp-open-rtos/examples/ESP-I2C-OLED$

sheinz, 2016-10-25:
Hi @Walter1604,

Are you sure you have function `user_init` defined in your code? It seems like the linker can't find it.
Function `user_init` is an entry point for user code execution.

UncleRus, 2016-10-25:
I just looked to this project (ESP-I2C-OLED) and it doesn't seem compatible with ESP Open RTOS.

sheinz, 2016-10-31:
Thanks @UncleRus. I will close the issue.

closed 2016-10-31 by sheinz