SuperHouse/esp-open-rtos #260: Crash in newlib with simple lwip sockets app

UncleRus, 2016-11-01:
**App:** https://gist.github.com/UncleRus/1d357810dbdc5b3b09e45807862be678
**How to reproduce:**
1. Build app & flash it
2. Connect to your esp using telnet: `telnet <ESP IP address> 80`
3. At the moment of connection app crashed in _mallinfo_r

**Stack trace:**
<details> 
```
Fatal exception (28):
epc1=0x40203afa

0x40203afa: _mallinfo_r at /home/gus/dev/esp/rtos/newlib/build/xtensa-lx106-elf/newlib/libc/stdlib/../../../../../newlib/libc/stdlib/nano-mallocr.c:497 (discriminator 2)

epc2=0x00000000
epc3=0x40209c45

0x40209c45: prvIdleTask at /home/rus/work/esp/esp-open-rtos/FreeRTOS/Source/tasks.c:4173

excvaddr=0x00000008
depc=0x00000000
excsave1=0x40203ae0

0x40203ae0: _mallinfo_r at /home/gus/dev/esp/rtos/newlib/build/xtensa-lx106-elf/newlib/libc/stdlib/../../../../../newlib/libc/stdlib/nano-mallocr.c:490

Registers:
a0 40203ae0 a1 3fff2b60 a2  00000030 a3  00005030

0x40203ae0: _mallinfo_r at /home/gus/dev/esp/rtos/newlib/build/xtensa-lx106-elf/newlib/libc/stdlib/../../../../../newlib/libc/stdlib/nano-mallocr.c:490

a4  3fff1448 a5  00000cd0 a6  00000008 a7  0000ceea
a8  00000002 a9  6a01000a a10 ad64bf16 a11 00000050
a12 3ffe92b0 a13 3fff2b80 SAR 0000001b

Stack: SP=0x3fff2b60
0x3fff2b60: 00000040 00000003 3fff2b80 40203a1c

0x40203a1c: mallinfo at /home/gus/dev/esp/rtos/newlib/build/xtensa-lx106-elf/newlib/libc/stdlib/../../../../../newlib/libc/stdlib/mstats.c:111

0x3fff2b70: 3ffebcb0 00000000 0000270f 4020a494

0x4020a494: xPortGetFreeHeapSize at /home/rus/work/esp/esp-open-rtos/FreeRTOS/Source/portable/esp8266/port.c:230

0x3fff2b80: 3feffe00 3ffea69c 3ffeb598 3fff2ca0
0x3fff2b90: 00000000 00000000 40100360 40100386

0x40100360: UserExceptionHandler at exception_vectors.o:?

0x40100386: UserExceptionHandler at exception_vectors.o:?

0x3fff2ba0: 40100000 40100fe6 00000030 40101016

0x40100000: _stext at ??:?

0x40100fe6: _xt_restore_interrupts at /home/rus/work/esp/esp-open-rtos/FreeRTOS/Source/portable/esp8266/port.c:230
 (inlined by) portENABLE_INTERRUPTS at /home/rus/work/esp/esp-open-rtos/FreeRTOS/Source/portable/esp8266/portmacro.h:173
 (inlined by) vPortExitCritical at /home/rus/work/esp/esp-open-rtos/FreeRTOS/Source/portable/esp8266/port.c:258

0x40101016: PendSV at /home/rus/work/esp/esp-open-rtos/FreeRTOS/Source/portable/esp8266/port.c:230

0x3fff2bb0: 00000028 3fff1434 3fff143c 4020cd84

0x4020cd84: tcp_alloc at /home/rus/work/esp/esp-open-rtos/lwip/lwip/src/core/tcp.c:1169

0x3fff2bc0: 3fff4a0c ffffffff 00001b8d 00000010
0x3fff2bd0: 3fff143c 3fff37a0 3fff1374 4021219c

0x4021219c: tcp_listen_input at /home/rus/work/esp/esp-open-rtos/lwip/lwip/src/core/tcp_in.c:466
 (inlined by) tcp_input at /home/rus/work/esp/esp-open-rtos/lwip/lwip/src/core/tcp_in.c:258

Fatal exception (28):
epc1=0x40203afa

0x40203afa: _mallinfo_r at /home/gus/dev/esp/rtos/newlib/build/xtensa-lx106-elf/newlib/libc/stdlib/../../../../../newlib/libc/stdlib/nano-mallocr.c:497 (discriminator 2)

epc2=0x00000000
epc3=0x40209c45

0x40209c45: prvIdleTask at /home/rus/work/esp/esp-open-rtos/FreeRTOS/Source/tasks.c:4173

excvaddr=0x00000008
depc=0x00000000
excsave1=0x40203ae0

0x40203ae0: _mallinfo_r at /home/gus/dev/esp/rtos/newlib/build/xtensa-lx106-elf/newlib/libc/stdlib/../../../../../newlib/libc/stdlib/nano-mallocr.c:490

Second fatal exception occured inside fatal exception handler. Can't continue.

ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 2292, room 16

0x40100000: _stext at ??:?
```
</details>

sheinz, 2016-11-01:
Hi @UncleRus,

There might be some issue with the semaphores, as removing them from the example fixes the crash.

UncleRus, 2016-11-01:
Semaphores issues will affect FatFs in multithreaded environment :(

sheinz, 2016-11-01:
I've just tested it on version before FreeRTOS upgrade and it doesn't crash with semaphores.
So, it seems like we have a regression :(

ourairquality, 2016-11-02:
It looks like the code could be wrong, taking the semaphore in the init thread without giving it back and then in the httpd_task thread - perhaps this leaves a bad state in FreeRTOS. Could you try rewriting it to keep the `started` flag in a static variable and use the semaphore to protect access to this, so take the semaphore and give it back in the same thread just while accessing this static variable.

UncleRus, 2016-11-02:
Taking semaphore in any thread without giving it back is absolutely normal for semaphores, I think.

ourairquality, 2016-11-02:
Quoting from the FreeRTOS documentation on the mutex: "This type of semaphore uses a priority inheritance mechanism so a task 'taking' a semaphore MUST ALWAYS 'give' the semaphore back once the semaphore it is no longer required." I re-wrote it as suggested and it seemed solid.

ourairquality, 2016-11-02:
Another clue might be that the `sdk_user_init_task` appears to exit and be deleted, and perhaps if it does so after taking the mutex but not giving it back then the FreeRTOS state become corrupted. Perhaps FreeRTOS should be more robust in this matter, but the code usage seems suspect too.

sheinz, 2016-11-02:
Thanks @ourairquality 

So, we too quickly jumped to a conclusion about regression. Luckily there's just mutex misuse.

I'll rename the issue so it tells more about the real problem

@UncleRus if everything is clear, you can close the issue.

closed 2016-11-02 by UncleRus