SuperHouse/esp-open-rtos #270: RFC: ms561101ba03 driver

mr-nice, 2016-11-11:
Hi all,
I hope adding an issue for this is OK. I have written a basic driver for ms561101ba03 barometic pressure sensor and want to ask for how to get it added to your project. You can find the source code here:

https://gitlab.com/mrnice/ms561101ba03

For now there are two drawbacks of the driver:

1. I am missusing vTaskDelay at ms561101ba03_get_temperature and ms561101ba03_get_pressure because it fits my needs. But I want to provide an async version of the two functions as well. But as I am new to FreeRTOS I am serching for the best way to provide one. Any comments are highly welcome!

2. I need to add the second order temperature compensation for higher accuracy. I will add that as soon as time permits.

I am happy for any comments on the code style etc. License is BSD but I can adjust the license if it is necessary. Even public domain would be OK for me.

After this I will start to write a driver for MPU-6000 gyroscope :)

Happy coding,
mr nice

sheinz, 2016-11-11:
Hi @mr-nice,

Thanks for getting in touch.

>I am missusing vTaskDelay at ms561101ba03_get_temperature and ms561101ba03_get_pressure because it fits my needs. But I want to provide an async version of the two functions as well. But as I am new to FreeRTOS I am serching for the best way to provide one. Any comments are highly welcome!

In my opinion, as the driver will be executed in context of a task, using vTaskDelay for delays that do not need precision is OK. It will allow other tasks to run while waiting.

One way to eliminate delays might be to split some functions into two. For example one to start measurement and another to read a measurement and relay that user will provide a delay between calls. But it will make driver usage a bit awkward and in my opinion not worth it.

> I am happy for any comments on the code style etc. License is BSD but I can adjust the license if it is necessary. Even public domain would be OK for me.

Coding style is described in the [README.md](https://github.com/SuperHouse/esp-open-rtos#coding-style)
BSD licence is OK.

I noticed that you use microseconds in vTaskDelay where milliseconds are expected:
```c
#define CONVERSION_TIME     20   // microseconds
vTaskDelay(CONVERSION_TIME / portTICK_PERIOD_MS);
```

Overall, the driver looks good to me. PR are welcome when you feel ready.
PS. Usually it is good to provide a small example with the driver.

mr-nice, 2016-12-03:
Hi all,
thanks for your time and comments. Sorry that I am a bit late with answering.
I updated the driver [1] and added an example [2]

My integer shifting and multiplication skills for mixed integer sizes and signed/unsigned got a bit rusty. So I now use constants and let the compiler do the shifting. Feel free to fix this :D

Nevertheless I tested the code from 5 to 27 degree C with reasonable results.

1 https://gitlab.com/mrnice/ms561101ba03/
2 https://gitlab.com/mrnice/ms561101ba03-example

closed 2017-01-28 by UncleRus