SuperHouse/esp-open-rtos #273: Clarification and possible bug in hw_timer

someburner, 2016-11-14:
I'm working on porting a onewire driver I made for the NONOS SDK that uses FRC1 to drive a state machine (and thus avoid blocking delays). The lowest time-step for Onewire (that I've seen) is 10us, so I use non-autoreload mode and set the timeout to 10us as needed to process read/writes.

In my NONOS code, I use clockdiv16 and the `US_TO_RTC_TIMER_TICKS(t)` macro provided by Espressif to set the value. I've tested this and it works pretty well.

I've set up a basic test with esp-open-rtos that looks like this:

**App code**:
```
void task1(void *pvParameters)
{
   while(1) {
      vTaskDelayMs(1000UL); // Print every second
      /* Should be close to 1000 */
      uint32_t count = getTestCount();
      printf("test count = %lu\n", count);
   }
}

void user_init(void)
{
   /* Init hw timer */
   hw_timer_init();
   hw_timer_start();

   xTaskCreate(task1, (signed char *)"hw_timer_task", 256, NULL, 2, NULL);
}
```

**Hw Timer code**:
```
static uint32_t testcout = 0;
static uint32_t loadval = 0;

uint32_t getTestCount(void)
{
   return testcout;
}

static void frc1_interrupt_handler(void)
{
   static uint32_t step_count = 0;
   /* Inc. every 10us * 100 = 1ms */
   if (++step_count == 100)
   {
      testcout++;
      step_count = 0;
   }
   timer_set_load(FRC1, loadval);
}

void hw_timer_init(void)
{
    /* set up ISRs */
    _xt_isr_attach(INUM_TIMER_FRC1, frc1_interrupt_handler);
}

void hw_timer_start(void)
{
   loadval = timer_time_to_count(FRC1, 10, TIMER_CLKDIV_16);
   printf("loadval = %u\n", loadval);

   timer_set_divider(FRC1, TIMER_CLKDIV_16);
   timer_set_reload(FRC1, false);
   timer_set_interrupts(FRC1, true);
   timer_set_run(FRC1, true);
}
```

Using timer_time_to_count with div16 I get this output:
```
loadval = 50
test count = 886
test count = 1779
test count = 2671
```
So testcount only gets to about 890 every time when I am expecting ~1000.

If I manually set loadval to 44 I get pretty close to 1000, +/- a couple us. 

I also tried the `timer_set_timeout()` method, but I was getting testcount's around ~700 or so. I looked around a bit and In `core/include/esp/timer.h` I saw this:
```
/* 
   ...
   Note that for a true "one shot" timeout with FRC1 then you need to
   also disable FRC1 in the timer interrupt handler by calling
   timer_set_run(TIMER_FRC1, false);
*/
int timer_set_timeout(const timer_frc_t frc, uint32_t us);
```

First, is `TIMER_FRC1` is supposed to just be `FRC1`? I get this error otherwise:

```
In file included from extras/hw_timer/hw_timer.c:10:0:
extras/hw_timer/hw_timer.c: In function 'frc1_interrupt_handler':
../../core/include/esp/timer_regs.h:15:28: error: incompatible type for argument 1 of 'timer_set_run'
 #define TIMER(i) (*(struct TIMER_REGS *)(TIMER_BASE + (i)*0x20))
                            ^
../../core/include/esp/timer_regs.h:16:20: note: in expansion of macro 'TIMER'
 #define TIMER_FRC1 TIMER(0)
                    ^
extras/hw_timer/hw_timer.c:93:18: note: in expansion of macro 'TIMER_FRC1'
    timer_set_run(TIMER_FRC1, false);
```

If I use `FRC1` to get it to compile, and set `timer_set_run(FRC1, false);` in the ISR, it seems to disable the interrupts (testcount stays at 0).

Also, tried all the same things but using `TIMER_CLKDIV_1` and with that I get testcount differences around 5500:

```
test count = 14
test count = 5435
test count = 10906
```

So yeah, any info on this would be helpful. Also I'm running the ESP @ 160MHz if that matters (didn't seem to change anything in NONOS, but that's using their macro for load value).

sheinz, 2016-11-14:
Hi @someburner,

If you need small precise delays in your code you can use `sdk_os_delay_us`. Please have a look at [i2c driver](https://github.com/SuperHouse/esp-open-rtos/blob/master/extras/i2c/i2c.c#L62) for example.

Regarding your example. If you want the interrupt handler to be called with fixed precise time periods you need to set the load value and set the timer to autoreload:
```c
void hw_timer_start(void)
{
   loadval = timer_time_to_count(FRC1, 10, TIMER_CLKDIV_16);
   printf("loadval = %u\n", loadval);

   timer_set_load(FRC1, loadval);
   timer_set_divider(FRC1, TIMER_CLKDIV_16);
   timer_set_reload(FRC1, true);
   timer_set_interrupts(FRC1, true);
   timer_set_run(FRC1, true);
}
```
I would also suggest placing interrupt handler into IRAM:
```c
static void IRAM frc1_interrupt_handler(void)
{
   static uint32_t step_count = 0;
   /* Inc. every 10us * 100 = 1ms */
   if (++step_count == 100)
   {
      testcout++;
      step_count = 0;
   }
}
```

The above modifications gives me pretty close count reading:
```
getTestCount: 8993
getTestCount: 9993
getTestCount: 10992
getTestCount: 11992
getTestCount: 12992
getTestCount: 13992
getTestCount: 14991
getTestCount: 15991
getTestCount: 16991
```

Another possible solution might be to [use already running timer FRC2 timer](https://github.com/SuperHouse/esp-open-rtos/blob/master/tests/cases/05_spiffs.c#L20)

someburner, 2016-11-14:
Thanks for the help. I was under the impression that `sdk_os_delay_us()` was a blocking delay. Is it not? 

Also, is `timer_set_reload()` an esp-open-rtos implementation? Or does that set the `FRC1_AUTO_LOAD` flag? I know esp-open-rtos is quite a bit different than Espressif's SDK, but the Espressif docs say that in auto-reload mode the acceptable range in us is 50 - x7FFFFF. But perhaps I'm confusing auto-load with reload..

I made the changes you suggested and it does indeed work. Thanks! For my implementation it's probably okay to use reload and then just enable/disable the timer, but I'm still a bit curious about this "true one-shot" comment.

Thanks again.

sheinz, 2016-11-14:
>Thanks for the help. I was under the impression that sdk_os_delay_us() was a blocking delay. Is it not?

`sdk_os_delay_us` delays by burning CPU cycles. As far as I understand it will not block the OS from task switching. If you really need a precision delay you need to disable OS task switching during the delay. For long delays that do not require a precision `vTaskDelay` is preferred way as it will directly say the OS to switch tasks right away.

> Also, is timer_set_reload() an esp-open-rtos implementation? Or does that set the FRC1_AUTO_LOAD flag? I know esp-open-rtos is quite a bit different than Espressif's SDK, but the Espressif docs say [...]

Hardware timer manipulation is [open source](https://github.com/SuperHouse/esp-open-rtos/blob/master/core/include/esp/timer.h), you can have a look.
```c
/* Write load value for timer. */
static inline void timer_set_load(const timer_frc_t frc, const uint32_t load)
{
    TIMER(frc).LOAD = load;
}
```
```c
/* Set timer auto-reload on or off */
static inline void timer_set_reload(const timer_frc_t frc, const bool reload)
{
    if (reload)
        TIMER(frc).CTRL |= TIMER_CTRL_RELOAD;
    else
        TIMER(frc).CTRL &= ~TIMER_CTRL_RELOAD;
}
```

someburner, 2016-11-15:
I don't need precision on the delays for this module, it's more that I plan to have other tasks running that I would like to give priority, and so I'm worried that using many delay_us() calls required for read/write operations would block for >1ms in some cases, which I'd like to avoid.

Yeah I looked into the source but the macros got a little cryptic for my level of experience with the project. But I think I have enough info to get me on my way for now. Thank you.

closed 2016-11-15 by someburner