SuperHouse/esp-open-rtos #282: LwIP v2 support

ourairquality, 2016-11-15:
LwIP v2 support is barely working now in the https://github.com/ourairquality/esp-open-rtos repo and patches are being submitted for inclusion in esp-open-rtos. This is now using a fork of the main LwIP git repo at https://github.com/ourairquality/lwip with only a small set of patches for esp-open-rtos.

The hope is that it will be easier to follow LwIP development which should be more productive than trying to backport fixes and this might give this project some more life. Perhaps some LwIP developers might even choose to use esp-open-rtos.

This also includes the wificfg example which is a quick test of some of the code paths.

There are some known issues with the DHCP server integration that need work, but a hack workaround is included.

Early testing and feedback welcomed.

pfalcon, 2016-11-15:
Great work! It is long on my todo list to do the same for OS-less SDK, so would be interesting to use your patch as a base for that. In that regard, a formal question - are your changes released under the same license (BSD license) as the original lwIP? Thanks!

ourairquality, 2016-11-15:
@pfalcon The LwIP work is based on the https://github.com/SuperHouse/esp-lwip.git The changes are small, almost trivial. Some I have written or re-written, but I would also be interested to know if there are any licensing issues with the LwIP patches. I might be useful to submit the patches upstream eventually.

As for the translation of some small parts of the SDK binaries to C code, I expect these are tainted, they retain a copyright notice. It will get more interesting when we rewrite some of these functions, and perhaps those could be split out into separate files with an open license.

projectgus, 2016-11-15:
The SuperHouse/esp-lwip repository is derived from kadamski's fork which is derived from a combination of an older Espressif source release and the LWIP upstream. None of the changes to LWIP modified its licensing (all LWIP files retain their original copyright and license notices), so there should be no issues with re-releasing or upstreaming.

closed 2017-10-07 by ourairquality