SuperHouse/esp-open-rtos #289: Travis CI broken

UncleRus, 2016-11-19:
Error:
`The command "openssl aes-256-cbc -K $encrypted_709d8233e262_key -iv $encrypted_709d8233e262_iv -in utils/travis_tests/sheinz_rsa.enc -out /tmp/sheinz_rsa -d" failed and exited with 1 during .`

sheinz, 2016-11-19:
Yes, I noticed :(
I explained the problem in https://github.com/SuperHouse/esp-open-rtos/pull/288
Unless I find a solution I will revert https://github.com/SuperHouse/esp-open-rtos/pull/272

sheinz, 2016-11-21:
Another solution might be to setup Jenkins CI on a test server and configure GitHub hooks to trigger a build there.

UncleRus, 2016-11-21:
@sheinz, That test server, is it your own server?

sheinz, 2016-11-21:
@UncleRus Yeah :) It's just a Raspberry PI with two esp8266 connected to it :)

UncleRus, 2016-11-21:
Cool :) 
I totally agreed with your suggestion about using Travis and Jenknins.

sheinz, 2016-11-21:
Ok.
As I'm not getting a reply from GitHub support regarding our issue. And I assume there's no an easy solution for it. I'll revert the PR https://github.com/SuperHouse/esp-open-rtos/pull/272 for now.

UncleRus, 2016-11-21:
How about spare test server? I also have the Raspberry and white IP :)

sheinz, 2016-11-21:
@UncleRus That would be really great!
We might need to think of a solution how to select a test server, or trigger a build on all of them...
I'll try to provide an instruction how to setup a Jenkins test server for esp8266 when I manage to do it on my side.

resetnow, 2016-11-21:
I also have a "server" (intel atom PC running 24/7). I'm behind NAT, though. I don't have spare modules but I can order them.