SuperHouse/esp-open-rtos #297: Optimization - unused tasks

SaimenSays, 2016-11-26:
Hi!
While searching for some RAM optimization/freeing I did also some investigation on system tasks. Two things I've recognized:

1. Can someone explain what `rtc_timer_task` and `pmTask` are for? In trace they seem never executed. Also there is no RTC on ESP?

2. `tftp_task` for OTA update. Why is this task created on init and not when packet arrives? I think it should be possible to register a udp binding and a callback with lwip. Maybe task can be created and ram allocated on demand, or it is not needed in the end. As it is now, task is only executed once on startup and then uses RAM but is mostly never used. Is there a reason why it is done like this?

Best Regards Simon

ourairquality, 2016-11-26:
@SaimenSays The `rtc_timer_task` is for the ets_timer, a high priority timer that seems to be used only for sending the beacons, so it might be possible to have it started only when beacons need to be sent. Would anyone know if the beacons need to be sent in a very accurate time slot, or might the use of the ets_timer for these be unnecessary?

The `pmTask` appears to be for power management. It's not clear if it could be avoided if not shutting down when idle. Might need to wait until we have some code for libpp.

The tftp server OTA does not appear very practical, but if using it on a secure network then perhaps there could be some trigger to enable it. Probably need OTA support that pulls the update from a server using http, with a signature check.

sheinz, 2016-11-27:
As @ourairquality explained `rtc_timer_task` is created in [ets_timer.c](https://github.com/SuperHouse/esp-open-rtos/pull/285/files#diff-b4e0d26e767b331f2a7704761d7de05dR304)
The name is really misleading. It has nothing to do with RTC. I think we don't need to preserve Espressif legacy naming and name it more sensibly.

If the timer value for sending beacons is multiple of 10 ms, than we can safely use FreeRTOS timer for it. And `est_timer.c` is not needed in the core.

Nevertheless I think hight precision timer might be useful. We might refactor `ets_timer.c` into some `extra` library that would provide such functionality. So, a user would include it only if it is needed.