SuperHouse/esp-open-rtos #298: sysparam negative int8 doesn't work

sheinz, 2016-11-27:
Retrieving negative value of int8 issues error `SYSPARAM_PARSEFAILED`

Test case:
```c
    status = sysparam_set_int8("int_2", -34);
    TEST_ASSERT_EQUAL_INT(status, SYSPARAM_OK);
    status = sysparam_get_int8("int_2", &int8_val);
    TEST_ASSERT_EQUAL_INT(status, SYSPARAM_OK);
    TEST_ASSERT_EQUAL_INT(-34, int8_val);
```

ourairquality, 2016-11-28:
@sheinz Nice to see some tests. fwiw the rewrite stores int8 and int32 as binary data rather than strings as this is generally more efficient and probably resolves this issue.

sheinz, 2016-11-28:
@ourairquality You are right. I missed the change that stores int8 as a single byte rather than int32.
So, this bug is actually fixed. Will close it when sysparam PR is merged.