SuperHouse/esp-open-rtos #306: RFC: software UART

mr-nice, 2016-12-04:
Hi all,
I started to port [1] esp8266-software-uart [2] to esp-open-rtos. I also added an example for how to use it here [3].

For now only RX is working, as I don't need more. I just wanted to inform you about it. Because it is a start and If any one else like to help I would be glad :)

It fits my needs for my project, but if there is interest in a full port and time permits I will do a full port.

I am not sure about if it is not possible to do the reading better then using sdk_os_delay_us? I am happy for any sugestions.

The license is MIT which should be compatible with BSD.

1 https://github.com/plieningerweb/esp8266-software-uart
2 https://gitlab.com/mrnice/esp-open-rtos-softuart
3 https://gitlab.com/mrnice/esp-open-rtos-softuart-example

UncleRus, 2016-12-04:
Please note that esp-open-rtos offers its own way of handling GPIO interrupts.

UncleRus, 2016-12-04:
Hi! It would be great if you created the pull request.
sdk_os_delay_us() just burns CPU cycles and good enough to use it in GPIO interrupt handler for delaying.

mr-nice, 2016-12-04:
I will add a pull request but I saw your interrupt handler merge just now :D I should start to pull more often. I will update the code now to use the new handler and add a pull request for it.

mr-nice, 2017-03-09:
Closed because it got merged.

closed 2017-03-09 by mr-nice