SuperHouse/esp-open-rtos #311: OTA

psankovic, 2016-12-10:
I've been trying to build ota example and run it but for some reason it's not working.
I was following OTA explanation from github wiki to the letter.
First I compiled esptool2 then rboot (with changed rboot.h) as wiki is saying.
I would like my application to be able to perform upgrade using OTA.
This is what I have done:
1. buil my app with Makefile having OTA=1 in it
2. clean flash using esptool.py -p /dev/tty.SLAB_USBtoUART erase_flash
3. load rboot.bin esptool.py -p /dev/tty.SLAB_USBtoUART --baud 115200 write_flash -fs 32m 0x0000 /esprtos/rboot/firmware/rboot.bin
4. load my bin to 0x2000 esptool.py -p /dev/tty.SLAB_USBtoUART --baud 115200 write_flash -fs 32m -fm qio -ff 40m 0x2000 mqtt_client.bin

ESP start to download new firmware from tftp but then it get stuck and get this:
TFTP client task starting...
Image will be saved in OTA slot 1.
Downloading mqtt_client.bin to slot 1...

 ets Jan  8 2013,rst cause:4, boot mode:(3,0)

wdt reset
load 0x40100000, len 1344, room 16 
tail 0
chksum 0xb2
load 0x3ffe8000, len 632, room 8 
tail 0
chksum 0x52
csum 0x52
csum err
ets_main.c 

Same thing is happening if I flash ota_basic example? What am I doing wrong? I am using esp-12e.

/regards,
     Petar

sheinz, 2016-12-13:
Hi @psankovic,

I think the wiki page is out of date.
You don't really need to do any additional steps. Simply flash the *ota_basic* example:
```bash
cd examples/ota_basic/
make flash
```
Make sure you provided your WiFi credentials so the device can connect to the AP.
This example provides two ways for OTA update:
 * Using built-in TFTP server
 * Using built-in TFTP client

First OTA update is the easiest. You simply push the firmware using TFTP client to the device:
```bash
tftp 192.168.2.2
tftp> binary
tftp> put firmware/ota_basic.bin firmware.bin
Sent 250240 bytes in 11.3 seconds
```
where 192.168.2.2 is the IP of the device and `firmware/ota_basic.bin` is the firmware image on local filesystem.

I haven't tried OTA update with built-in TFTP client.

sheinz, 2016-12-13:
I've updated the wiki page.

psankovic, 2016-12-13:
Thanks for help. I finally have tftp server running on ESP and I'm able to flash it. However if ESP is tftp client then I can not connect to server. 
I tried to debug it and what I realised is that code get stuck at: tftp_receive_data(). Any chance you could try it as well?
Petar

sheinz, 2016-12-14:
Hi @psankovic,

I tried the variant when ESP is a TFTP client and it kinda works...
The following modifications of the example is necessary:
 * IP of TFTP server. The local machine with TFTP server running.
 * SHA256 of the firmware.bin or comment out SHA validation.
 * Run TFTP server on port 70 (notice `TFTP_PORT+1` in the code) or comment out running TFTP server on port 69 and change `TFTP_PORT+1` to `TFTP_PORT`.

You can check if ESP even attempts to download a firmware with tcpdump:
```bash
tcpdump -i bridge100 port tftp
```
where *bridge100* is your interface with ESP

I commented out SHA validation and my ESP downloads a firmware, writes it and restarts (restart not always works for some reason)
```
Image addresses in flash:
 0: offset 0x00002000
*1: offset 0x00102000
mode : sta(18:fe:34:ca:da:b5)
add if0
TFTP client task starting...
Image will be saved in OTA slot 0.
Downloading firmware1.bin to slot 0...
ota_tftp_download firmware1.bin result -4
scandone
add 0
aid 1
cnt

connected with esp-test, channel 3
dhcp client start...
ip:192.168.2.2,mask:255.255.255.0,gw:192.168.2.1
Downloading firmware2.bin to slot 0...
ota_tftp_download firmware2.bin result 0
Downloading firmware1.bin to slot 0...
ota_tftp_download firmware1.bin result 0
Looks valid, calculating SHA256...
Image SHA256 = a64e6ab237fa78a0d4c89c72bd4c57b658b60ed868657760afb4e21e7575cfeb
SHA256 Matches. Rebooting into slot 0...
```

The *ota_basic* example needs to be reworked. It requires to much hassle just to try it out.

mehdilauters, 2016-12-14:
Hi,
talking about OTA, I am trying it for several days based on the freeRTOS sdk without really success.
I spent a lot of time just trying to find out which output file had to be sent as I was looking for changes in the build folder. I think that it should be written somewhere into the wiki OTA page that the file real OTA firmware is firmware/PROGRAM.bin.

I saw that there were targets ``FW_FILE_1`` and ``FW_FILE_2`` but how is it possible to generate several rom at different flash addresses without changing the ``ld/program.ld`` file?

As if I guess (or I did not succeed trying it: FW_FILE_2 was never bootable), it is not, I will try to find a correct solution in order to be able to generate both roms with correct linkage and put it in a merge request.

I finally get it working by manually changing the address in ``ld/program.ld`` file at line `` irom0_0_seg : org = 0x40202010, len = (1M - 0x2010)`` adapting it for each rom, but is is the current official solution?

Mehdi

sheinz, 2016-12-14:
Hi @mehdilauters,

ESP can address only 1MB of program memory. Current implementation of OTA update supports only so called "big flash" mode where each firmware is placed at 1MB aligned start. In that case only one firmware binary is needed as addresses will be the same for both firmwares.

For example popular modules ESP12-E have 4MB flash size. So, the first firmware will be located just at flash start. The second firmware - at offset 0x100000 (1MB). So, ESP12-E can contain up to 4 firmwares at the same time.

Only one firmware image is needed for OTA, which is placed in `firmware/<PROGRAM>.bin`

mehdilauters, 2016-12-14:
Thank you for your answer.
So what are targets ``FW_FILE_1`` and ``FW_FILE_2`` for?

sheinz, 2016-12-14:
It is for the case when the bootloader (rboot) is not used. Currently bootloader is used by default.

From esptool documentation:
>The default command output is two binary files: my_app.elf-0x00000.bin and my_app.elf-0x40000.bin. You can alter the firmware file name prefix using the --output/-o option.

>elf2image can also produce a "version 2" image file suitable for use with a software bootloader stub such as rboot or the Espressif bootloader program. You can't flash a "version 2" image without also flashing a suitable bootloader.

>esptool.py elf2image --version=2 -o my_app-ota.bin my_app.elf

projectgus, 2016-12-14:
> restarts (restart not always works for some reason)

This might be totally off-base, but when running immediately after flashing with esptool.py, software restarting will send you back to the serial ROM bootloader (because the strapping pin register is loaded on hardware reset and doesn't change, so it sees GPIO0=LOW). So the first "OTA" update after a serial flash will always reset to the bootloader.

> > So what are targets FW_FILE_1 and FW_FILE_2 for?
> It is for the case when the bootloader (rboot) is not used. Currently bootloader is used by default.

I think these are probably cruft at this point, that I forgot to take out (sorry!)

The Makefile used to support an "OTA mode" which used rboot and a "non-OTA" mode which didn't, but since d9202af2 the "non-OTA" mode was removed and rboot is used for all cases, to simplify things.

> The ota_basic example needs to be reworked. It requires to much hassle just to try it out.

I agree. FWIW, I added the SHA validation because I didn't like having a totally insecure OTA as the default example, and using SHA at least validates the image - even if it's readable in transit, it can't be spoofed as long as you read the SHA over a secure connection. HTTPS OTA would be infinitely preferable!

salehjg, 2016-12-24:
hi
im trying to test "ota_basic" example on an ESP module with 512KB dataflash but somehow it fails and causes wdt reset.
i've used Advance tftp as server(On Ubuntu) and ESP as client.

i also modified  IP Address and Port (to 69) and changed FLASH_SIZE to 4 (parameters.mk) and rebuilt bootlader directory with 'make bootloader' as mentioned in readme file.

1- anyway is it even possible to use OTA while dataflash's capacity is less than 1MB ?
2- is it possible to specify size of ROM slots?

-------
by the way sorry for poor English :)

psankovic, 2016-12-27:
Hello,

After letting this OTA problem to soak for a while, I decided it's time to return to it. I know that this is not best place to report "user" problem, but with lack of forum for esp-open-rtos I decided to ask again for help here. I have TFTP client running on ESP and TFTP server running on local comp. Now after a lot of fiddling I have my client requesting firmware from server and download is starts. After a while ACK's from ESP are not sent any more and WDT restart ESP. Usually it happens after ~40 packages of data transfer. 
What I noticed is that it get stuck in this if statement in ota-tftp.c

        /* Reset retry count if we got valid data */
        retries = TFTP_TIMEOUT_RETRANSMITS;

        if(write_offs % SECTOR_SIZE == 0) {
            sdk_spi_flash_erase_sector(write_offs / SECTOR_SIZE);
        }

Initially I flash my ESP with this command (directly from eclipse)

/esprtos/esp-open-sdk/esptool/esptool.py -p /dev/tty.SLAB_USBtoUART --baud 460800 write_flash -fs 16m -fm qio -ff 40m 0x0 ../../bootloader/firmware_prebuilt/rboot.bin 0x1000 ../../bootloader/firmware_prebuilt/blank_config.bin 0x2000 ./firmware/mqtt_client.bin

Can anybody suggest what to do?
/regards,
    Petar

phlemoine, 2016-12-27:
Could this be related to #151 https://github.com/SuperHouse/esp-open-rtos/issues/151 ?

psankovic, 2016-12-27:
I do not believe since #151 refers to TCP traffic and Tftp is UDP. 
Petar

psankovic, 2017-01-29:
Hello,

Wanted to share my findings with community about OTA. After a lot of struggling eventually I gave up on OTA. It was simply not working in my case. Today I returned to it again, but due to some other problems couple days ago I have added capacitors on power pins. And guess what, OTA is working like a charm. 

Thought of the day: PUT STUPID CAPACITORS ON POWER PIN it will save you a lot of struggling even though you think you do not need them, put them anyway.
Thanks to all for help in this forum.
/regards,

mikaelsundin, 2017-01-30:
Yes its a common problem. Sporadic restart can also be a sign or sensetive for EMI,ESD..

kanflo, 2017-04-12:
Glad you sorted it out @psankovic. Feel free to open another issue if you run into problems with OTA.

closed 2017-04-12 by kanflo