SuperHouse/esp-open-rtos #312: private libraries?

malachib, 2016-12-16:
I'm wondering is there a way to use symbolic linking to pull in an extra chunk of source code?  Specifically:

```
+ project root
  - main.c
  - Makefile
  + FreeRTOS-Plus-CLI (symbolic link to elsewhere)
   - FreeRTOS_CLI.c
   - FreeRTOS_CLI.h
```

I do this all the time with my PlatformIO projects and it's quite handy.  I'm a little intimidated by upgrading other people's projects to full components.  For the record, when I try to use PROGRAM_SRC_DIR for this, I get:

`/home/malachi/Projects/playground.freertos/ext/FreeRTOS-Plus/FreeRTOS-Plus-CLI/FreeRTOS_CLI.c:1: *** missing separator.  Stop.`

Incidentally, this esp-open-rtos is pretty awesome!

sheinz, 2016-12-16:
Hi @malachib,

I'm not sure about symbolic links but you can use git submodules for extra components that are used in multiple projects.
Have a look how I used it [here](https://github.com/sheinz/esp-midea-ir-demo)

malachib, 2016-12-16:
@sheinz thanks, that helps demystify component-izing libs.  Not so bad after all.

As much as I love git submodules (no sarcasm, I really do love em) - symbolic linking would be way better at the moment

malachib, 2016-12-16:
Turns out EXTRA_COMPONENTS does work with symbolic linking.  This solves my issue, thank you  @sheinz for your help!

malachib, 2016-12-16:
I spoke too soon.  EXTRA_COMPONENTS will reach out to a symbolic link place, but the component.mk inside will not, which unfortunately still causes me problems.

closed 2017-01-06 by malachib